esp8266/Arduino #823: Flash Size Mapping (OTA+SPIFFS) and sketch size

hallard, 2015-09-27:
Hi guru's 
I've got a question regarding flash mapping,let's say I want to use OTA and SPIFFS and I got 1MB flash size.
Assuming I'm using OTA so I can't select more than 512KB (correct ?)
So I'm selecting 512K (64K SPIFFS), this mean my sketch maximum size should not go over 512-64 ? Is that correct ?
Thank you very much for any clarification

Links2004, 2015-09-27:
That OTA is possible you need 2 times the sketch size. you need to select 1MB but your code shut not go over ~512Kb or the OTA will give you an error.

hallard, 2015-09-27:
@Links2004 
yes, that what I understood, thanks. 
I've got 1MB flash and I munimal SPIFFS is 64KB, 512KB (64KB SPISSF) in Arduino Menu, does this mean my maximum sketch size is 512-64 so 448KB ?

Links2004, 2015-09-27:
SPIFFS  is only ones in but there also some other areas the SPIFFS in the flash.
so its around 450KB.
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/reference.md#file-system

hallard, 2015-09-27:
@Links2004 
Correct I didn't mention the other areas ;-)
As soon as the linker do the job to not to oversize this around 450Kb with a compilation error, it's fine for me, thank for your explanation.

closed 2015-09-28 by igrr