esp8266/Arduino #824: Compiling HTTPSRequest Example Error

jarek319, 2015-09-27:
I'm not sure if the issue is with the method I used, or the library itself, but here is what I did:
1. Downloaded Arduino 1.6.5r5 from Arduino.cc and installed
2. Added ESP8266 Staging version to Arduino board manager and installed ESP8266 core using Board manager
3. Closed Arduino IDE
4. Downloaded this as a zip and unzipped contents of Arduino-esp8266.zip\Arduino-esp8266\hardware\esp8266com\esp8266\libraries\ESP8266WiFi into C:\Users\jarek\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-1106-g8253b82\libraries\ESP8266WiFi
5. Opened Arduino IDE and opened HTTPSRequest Example and hit compile

The output I got was: 

C:\Users\jarek\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-1106-g8253b82\cores\esp8266\libc_replacements.c:408:22: warning: 'struct tm' declared inside parameter list [enabled by default]
 time_t mktime(struct tm *timp) {
                      ^
C:\Users\jarek\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-1106-g8253b82\cores\esp8266\libc_replacements.c:408:22: warning: its scope is only this definition or declaration, which is probably not what you want [enabled by default]
ESP8266WiFi\WiFiClientSecure.cpp.o: In function `WiFiClientSecure::operator=(WiFiClientSecure const&)':
C:\Users\jarek\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-1106-g8253b82\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp:186: undefined reference to `ssl_match_fingerprint'
ESP8266WiFi\WiFiClientSecure.cpp.o: In function `cbuf':
C:\Users\jarek\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-1106-g8253b82\cores\esp8266/cbuf.h:28: undefined reference to `ssl_match_fingerprint'
collect2.exe: error: ld returned 1 exit status
Error compiling.

I tried cloning the git and building it myself but got even more errors that way (4 errors in build.xml, all related to windows-specific build variables).

jarek319, 2015-09-27:
P.S. for some reason immediately before the ssl_match_fingerprint text in the error message above, there are these things ` instead of ' quotes. Not sure why.

igrr, 2015-09-28:
Duplicate of #740.

igrr, 2015-09-28:
Oh, sorry, i didn't notice the part with `undefined reference tossl_match_fingerprint`.
Apparently you didn't upgrade your copy of libaxtls.a. It's not part of ESP8266WiFi library, you have too look for it inside `hardware/esp8266com/esp8266/tools/sdk/lib/libaxtls.a`.

closed 2015-09-28 by igrr

jarek319, 2015-09-29:
Thank you! That did it :) I copied the newer libaxtls.a file from the zip and replaced the one at the path you mentioned. Working great now, thanks!