esp8266/Arduino #825: There are a new product WiFi-MCU:STM32F4+WiFi(Broadcom)+Lua

asetyde, 2015-09-27:
It can be in future have support ?

It seems a real son of NodeMCU but more power , gpio and solid product.

I ask to you because this community is strongly efficient and is more live than other to upgrade add and improve .

https://www.kickstarter.com/projects/857552561/wifi-mcu-stm32f4-wifibroadcom-lua

EUA, 2015-09-27:
It is not related with this project.
It's also pricey ($8). You can buy ESP for $2.
But I wish I start development with it that one due INSTABILITY of esp8266.
With esp8266, you need to learn o live with Random Resets, bootloops etc... :(

Duality4Y, 2015-09-27:
the esp8266 is very stable, when you start programming it with the sdk.

asetyde, 2015-09-27:
You can use nodemcu , 6$. And good level of usability

asetyde, 2015-09-27:
This question is for big developer of this board esp . How are you  idea ? @Ivan

Links2004, 2015-09-27:
the hardware concept look complete different to the ESP.
they have an 2M flash + cortex M4 (STM) + Broadcom wifi chip (all connected by SPI).

the good points i can find:
- registers for the STM32 are documented.
- the Broadcom may handle the basic WiFi level may less IRQ problems.
- debugger can be used (Cortex)

bad thinks:
- the official git was deleted (https://github.com/MXCHIP/MICO), but the WiFi part there most likely also blobs.
- bare module costs 4 times more
- the 2M flash is not for code (512K ship flash only)
- same problem of documentation then the ESP (excepted the stuff form STM direct)

support inside the esp8266 arduino code is impossible,
to many differences other compiler, registers.

for the moment I think its easier and cheaper to attache a arduino mini pro to the ESP if more IOs / ADC or HW PWM etc is needed.

asetyde, 2015-09-27:
Thanks for analisys Markus , i also see many of this good/evil points . 
I prefer ESP and this strong community ! 

thanks a lot

igrr, 2015-09-28:
Basically it's a completely different product. 
You can run Arduino code on the host processor already (check out stm32duino.com), and writing a variant of ESP8266WiFi library to communicate with Broadcomm chip over SPI should also be straightforward. 

Regarding stability, well, your mileage with ESP will vary depending on how you use it. I've got a bunch of sensors running unattended for months now, and they only fail when power goes out. If you use static memory allocation, avoid long interrupt-locking routines, and understand the limitations of certain Arduino APIs, then you will be fine. Obviously there are major issues with the core itself (like the now-resolved issues with OTA updates failing, or the heap corruption issue). If you want something more reliable, go for a *product* like Digistump Oak.

closed 2015-09-28 by igrr

asetyde, 2015-09-28:
Thanks Ivan ! ESP is my best choice at now.

hallard, 2015-09-28:
@igrr 
You tested OAK ?!?! I'm waiting for mine and I'm really impatient to know how we can develop for it (Arduino ESP8266, specific IDE, Particle ) Still very mysterious ...