esp8266/Arduino #828: SPIFFS Upload failed on OSX

luc-github, 2015-09-28:
Hi when I do not have any issue under windows 7 - I have an error when uploading data content under OSX, but uploading sketch itself is ok.

```
Arduino : 1.6.5 (Mac OS X), Carte : "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS)"

[SPIFFS] data   : /Users/luc/ESP8266/esp8266/data
[SPIFFS] size   : 64
[SPIFFS] page   : 256
[SPIFFS] block  : 4096
/gpl.txt
[SPIFFS] upload : /var/folders/h9/xf835jyx4v76smg_d95l_45m0000gn/T/build5852857672375480751.tmp/esp8266.spiffs.bin
[SPIFFS] reset  : ck
[SPIFFS] port   : /dev/cu.usbserial
[SPIFFS] speed  : 115200
[SPIFFS] address: 0x6B000

Uploading 65536 bytes from /var/folders/h9/xf835jyx4v76smg_d95l_45m0000gn/T/build5852857672375480751.tmp/esp8266.spiffs.bin to flash at 0x0006B000
warning: espcomm_send_command: no final C0
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
SPIFFS Upload failed!
```

is there something that need to be done on OSX compare to Windows ?

Thanks

igrr, 2015-09-28:
Did you use a [PL2303-based adapter](https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/boards.md#serial-adapter) to upload? Does the same happen when you upload the sketch itself?

luc-github, 2015-09-28:
Yes I used PL2303 - I missed this part on read me - but upload sketch is working using same adapter - did not got any error

igrr, 2015-09-28:
Okay, I'll leave a link to the related esptool-ck issue here and close this one.
https://github.com/igrr/esptool-ck/issues/9

closed 2015-09-28 by igrr

luc-github, 2015-09-28:
Ok I did not thinked about adapter issue as upload sketch was ok - thank you