esp8266/Arduino #829: Is possible record the firmware?

alexandre-mbm, 2015-09-28:
Sorry because this is not the project's issue. But I do not find more expert than this site.

I know that I can load sketches on ESP8266 through Arduino IDE. I also know that I can do AVR flashing using a Arduino Uno or Nano as Arduino ISP through Arduino IDE. There are specifics tutorials for these.

I would to know if (and how) I could do flashing on ESP8266 from an Arduino as ISP.

I'm interested in changing the FTDI converter by an Arduino Nano.

alexandre-mbm, 2015-09-28:
If I use TX/RX of a SparkFun's LLC, for 3.3 V, perhaps it works with SDIO, SPI or I²C. But this do not implies it have UART support, is not?

mangelajo, 2015-09-30:
You could use a voltage divider on the 5V TX uart to the 3.3V RX on ESP8266, also a simple resistor in series (1k) may work, due to the contention of the latch-up protection diode inside the ESP8266 chip. (which will limit the voltage to 3.3V + Vd)

alexandre-mbm, 2015-09-30:
Let's see if I understand. These solutions do not cause ripple because of a latch-up protection diode that there is inside the ESP8266 chip.

devyte, 2017-10-20:
Closing per #3655 .

closed 2017-10-20 by devyte