esp8266/Arduino #830: Client closing an HTTP form upload crashes the server.

mangelajo, 2015-09-28:
I have found this while testing the web update example, if you cancel the upload of the new binary from server, and then you try to upload a new one, you get the connection open and accepted, but then it crashes on:

LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1

I suspect the loop on ESP8266WebServer::_parseForm is not handling client disconnection, I tried to add a if (!client.connected()) return false inside the while(1) loop without success...

mangelajo, 2015-09-28:
Hmm, ok, the reason is that https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/src/Parsing.cpp#L277  uploadReadByte will lock forever in the while loop.

We should check when client.connected() goes false, and then send a UPLOAD_FILE_ABORTED or similar to the handler, so the app would be able to cleanup.

Links2004, 2015-09-28:
can you try: 
```cpp
uint8_t ESP8266WebServer::_uploadReadByte(WiFiClient& client){
  int res = client.read();
  if(res == -1){
    while(!client.available()) {
      yield();
      if(!client.connected()) {
          break;
      }
    }
    res = client.read();
  }
  return (uint8_t)res;
}
```

mangelajo, 2015-09-28:
I will send you a patch, that works, but also you have to make the loop on top of this (_parseForm) check client.connected() and exit when necessary.

mangelajo, 2015-09-29:
I sent a pull request with the patch. It's been manually tested.

closed 2015-11-30 by igrr