esp8266/Arduino #832: cannot open linker script file eagle.flash.4m1m.ld: No such file or directory

zenmanenergy, 2015-09-28:
On my nodemcu 12e if I set a flash size of 4M (1M spiffs) I get the following error. This doesn't happen if I choose a 3M spiffs size:

/Users/stevenelson/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1106-g8253b82/cores/esp8266/libc_replacements.c:408:22: warning: 'struct tm' declared inside parameter list [enabled by default]
time_t mktime(struct tm *timp) {
^
/Users/stevenelson/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1106-g8253b82/cores/esp8266/libc_replacements.c:408:22: warning: its scope is only this definition or declaration, which is probably not what you want [enabled by default]
/Users/stevenelson/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot open linker script file eagle.flash.4m1m.ld: No such file or directory
collect2: error: ld returned 1 exit status
Error compiling.

igrr, 2015-09-28:
Fixed in git (#765), but staging version has not been updated yet.

zenmanenergy, 2015-09-28:
Wow, talk about fast turn around time! Ivan if you lived near me I would owe you so many beers! Thanks man.

closed 2015-09-28 by zenmanenergy

Juppit, 2015-09-28:
> /Users/stevenelson/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1106-g8253b82/cores/esp8266/libc_replacements.c:408:22: warning: 'struct tm' declared inside parameter list [enabled by default]
> time_t mktime(struct tm *timp) {

this issue is not fixed afaik.

But @igrr wrote in #740:
> Thanks, I'm aware of that. You can safely ignore this warning, I'll fix it along with the next round of TLS-related stuff.

On 28.09.2015 at 18:39 wrote Ivan Grokhotkov:

igrr, 2015-09-28:
I was referring to `cannot open linker script file eagle.flash.4m1m.ld: No such file or directory` — that one was fixed in git.

mktime stuff is just a warning and may be safely ignored.