esp8266/Arduino #835: Is there any way to reduce rf power?

masaorobotics, 2015-09-28:
While taking readings with another analog microcontroller I need the esp transmit that data continuously, everything works fine, except that the amount of noise induced on my analog input is quite high.

I've been a level of excellence :) power supply, also uncoupling, filtering, chokes etc. Obviously each microcontroller with independent feed sources (but invariably will be physically close to each other)

There is a way to temporarily cut off the transmission briefly? without resorting to a "deep sleep" ???

It is my first time writing something down here and apologize in advance if I'm doing it in place or in the wrong way.

Greetings and thank you very much!
Paul, Greetings from Argentina.

olivernz, 2015-09-28:
Haven't used it but the doco says:

   - WiFi.mode(m): set mode to WIFI_AP, WIFI_STA, WIFI_AP_STA or *WIFI_OFF*.

So it must be able to be switched off with the module still going.

masaorobotics, 2015-09-28:
I understand, the problem is that I take readings every five seconds, the process of taking those readings delay about 2 seconds.
Once the reading taken I transmit to esp and esp should immediately send it to the network. The problem is that if I turn off wifi, esp is disconnected from the network .. for when I return to turn the wifi esp after connecting with SSID and password, the elapsed time is much to my application.-

The esp have much power and powerful what is good ... but when we have 8 esp in a room? OMG

On the other hand, after this project, I am very excited to develop a VO-IP application, something like "Zello esp" :) with esp, where while taking readings the transmit virtually in real time, so this will be essential to take the controlling the RF power, unless there is another way to reduce noise.

Oliver, I am grateful for your response and time.
Greetings!
Paul!

masaorobotics, 2015-09-28:
Thank you again, I see that the time it takes to connect to the network depends greatly between different brands and models of routers. I also believe that the power control later or rather sooner than later will be a necessity, clarified that the esp know if the Ic actually able to control that.
Paul

RussMathis, 2015-09-29:
I have not had time to try this feature, let me know if it works:

void system_phy_set_max_tpw(uint8 max_tpw)
options as follow: Unable to find enumerations for these settings so the L-Values follow;

    0-82: Sets the maximum value of RF TX power, units: 0.25dBm

asetyde, 2015-09-29:
Why reduce it ?

RussMathis, 2015-09-29:
@asetyde, perhaps reducing db before reading adc ripple. :)

asetyde, 2015-09-30:
It can be used to extend or not power signal in AP mode

asetyde, 2015-12-01:
Please close this issue , not other to discuss

closed 2015-12-03 by masaorobotics