esp8266/Arduino #838: ESP8266 does not honour setNoDelay(true)

drmpf, 2015-09-29:
My sketch has this code
bool NoDelayFlag = true;
...  
server.begin();
server.setNoDelay(NoDelayFlag);

A detailed examination of the wireshark capture shows when sending the response
{.<prompt not set>`6000|C~Toggle Button}
The first packet has just { and the second packet has the rest BUT
the second packet is delayed by 200mS until the first packet is acked 
This should not happen with setNoDelay(true)

CSV of packet details
,Time Of Day,Seq,Source,Destination,Len,Hdr,Net,Tspt,Flags,Data,Blks,Function,Content,Comment
,18:38:58.175649,678,ESP8266: 4989,DELL:50561,64,58,IP4,TCP,   AP   ,1,,,{,1 byte over
,18:38:58.396734,679,DELL:50561,ESP8266: 4989,64,58,IP4 DF,TCP,   A    ,,,,,
,18:38:58.408653,680,ESP8266: 4989,DELL:50561,97,58,IP4,TCP,   AP   ,39,1,,.<prompt not set>`6000|C~Toggle Button},40 bytes aggregated\ Trans: Request in 0.2418 secs

Links2004, 2015-09-29:
the setNoDelay is only valid for the listening socket if a new client connects you get a new socket.
(returned by available) 
for the new socket (WiFiClient) you need to set setNoDelay again.

drmpf, 2015-09-29:
Thanks I will try that,
does
``` 
void WiFiServer::setNoDelay(bool nodelay){
  if(!_pcb) return;
  if(nodelay) tcp_nagle_disable(_pcb);
  else tcp_nagle_enable(_pcb);
}
```
do anything then or should it be removed from the code?

Links2004, 2015-09-30:
I shut be changed, currently its apply the NoDelay setting to the listening socket witch make not the most sense since this socket never sends data ;)
may storage a flag for the mode and then apply setNoDelay to the client when the new socket is created.

igrr, 2015-09-30:
I'm not sure this should be changed. Reading lwip source, this is the
correct way to disable Nagle algorithm. Need to test the case reported by
@drmpf and see what is causing the issue.

@drmpf can you please provide the sketch you used to test? Thanks.

drmpf, 2015-10-01:
Hi iggr,
this reproduces the problem, second byte from ESP held up until get ack back from client after 200mS

```
#include <EEPROM.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>

// normally DEBUG is commented out
#define DEBUG

unsigned long timeout = 15000; //15 sec
bool NoDelayFlag = false;  // normally false, only set to true for telnet use
WiFiServer server(80);
WiFiClient client;

void setup ( void ) {
  WiFi.mode(WIFI_STA);
  EEPROM.begin(512);
  Serial.begin (9600);
  delay(10);
  Serial.println();
  for (int i = 4; i > 0; i--) {
#ifdef DEBUG
    Serial.print(i);
    Serial.print(' ');
#endif
    delay(500);
  }
  char ssid[] = "xxxxxx"; 
  char password[] ="xxxxxx";
  char staticIP[] ="10.1.1.100";
  uint16_t portNo = 4989;
  server = WiFiServer(portNo);
  // Initialise wifi module
#ifdef DEBUG
  Serial.println(F("Connecting to AP"));
  Serial.print("ssid '");
  Serial.print(ssid);
  Serial.println("'");
  Serial.print("password '");
  Serial.print(password);
  Serial.println("'");
#endif
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
#ifdef DEBUG
    Serial.print(".");
#endif
  }
#ifdef DEBUG
  Serial.println();
  Serial.println(F("Connected!"));
#endif

  if (*staticIP != '\0') {
    // config static IP
    IPAddress ip(10,1,1,100);
    IPAddress gateway(10,1,1,1); // set gatway to ... 1
#ifdef DEBUG
    Serial.print(F("Setting gateway to: "));
    Serial.println(gateway);
#endif
    IPAddress subnet(255, 255, 255, 0);
    WiFi.config(ip, gateway, subnet);
  } // else leave as DHCP

  // Start listening for connections
#ifdef DEBUG
  Serial.println(F("Start Server"));
#endif
  server.begin();
  server.setNoDelay(NoDelayFlag);
#ifdef DEBUG
  Serial.println(F("Server Started"));
  // Print the IP address
  Serial.print(WiFi.localIP());
  Serial.print(':');
  Serial.println(portNo);
  Serial.println(F("Listening for connections..."));
#endif

  client = server.available();
#ifdef DEBUG
  Serial.print("+++"); // end of setup start listening now
#endif
}

unsigned long timeoutTimerStart = 0;

bool alreadyConnected = false;
// the loop routine runs over and over again forever:
void loop() {

  if (!client) { // see if a client is available
    client = server.available(); // evaluates to false if no connection
  } else {
    // have client
    if (!client.connected()) {
      if (alreadyConnected) {
        // client closed so clean up
        closeConnection();
      }
    } else {
      // have connected client
      if (!alreadyConnected) {
#ifdef DEBUG
        Serial.println("ConnectionOpened");
#endif
    client.setNoDelay(NoDelayFlag); // << this does not make any difference
        alreadyConnected = true;
        // start timer
        timeoutTimerStart = millis();
      }
    }
  }

  if (client) {
    if (client.available()) {
      // got some msg
      // send back {.}
      while (client.available()) {
        client.read();
      }
      delay(10);
      client.write("{");
      //delay(3);
      client.write(".");
      client.write("}");
    }
  }
  // see if we should drop the connection
  if (alreadyConnected && (timeout > 0) && ((millis() - timeoutTimerStart) > timeout)) {
    closeConnection(); 
  }
}

void closeConnection() {
#ifdef DEBUG
  Serial.println("ConnectionClosed");
#endif
  alreadyConnected = false;
  if (!client) {
    return;
  } // else
  client.stop();
  client = server.available(); // evaluates to false if no connection
}

```

drmpf, 2015-10-01:
p.s. this was compiled under esp8266-1.6.5-947-g39819f0
Also tried setNoDelay(true) on the client object but no difference.

drmpf, 2015-10-04:
@igrr Found this possible reason on
https://github.com/jeelabs/esp-link/issues/46

```
What happens is that the SDK pretty much only supports a single packet in flight (it's difficult to get LwIP to do more). So you're limited to the round-trip time data->ACK at the TCP level.
```
So if I send just one byte in a packet then the ESP will not send anything else until that packet is ACKed which is what I am observing,  not great but...

This may also explain some complaints about slow webpages, if many small packets are sent and the acks take 0.2sec each

p.s. on another topic would you please add a note to the reference page indicating how much stack the Arduino code (loop() ) has available and the total size of global vars that can be allocated from the Arduino sketch.

drmpf, 2015-10-13:
@igrr  Perhaps we can just fix this in the documentation for setTCPNoDelay() method, by noting regardless of this setting only one packet at time is on the wire.

igrr, 2016-03-11:
Turns out, WiFiServer::setNoDelay resulted in undefined behaviour.

igrr, 2016-06-23:
Fixed in git version. Now we have proper behaviour for setNoDelay, both for WiFiClient and WiFiServer.

closed 2018-01-02 by igrr