esp8266/Arduino #841: Audio playback

BossensM, 2015-09-30:
Is there any support for audio playback? I get compile errors while trying to use Audio.h

mangelajo, 2015-09-30:
I guess that requires DAC support, we're supposed to be able to set the I2S module in a "DAC" pwm which you can filter through a low pass filter. (At least that's what I understood from the mp3 example Espressif provides).

tim-in-oakton, 2015-10-23:
I did a project last year with my daughter that used a high freq PWM to generate sound (using a WAV, not MP3) - no low pass required:
http://www.nova-labs.org/blog/2014/02/26/tardis-pinewood-derby-car-with-light-and-sound/ 

note - this used a teensy, not an ESP - but the approach should work as long as the PWM libs work

devyte, 2017-10-20:
The I2S api covers this. It's a bit primitive (i.e.: C-style and not a class), but it's reported to work.
Closing.

closed 2017-10-20 by devyte