esp8266/Arduino #844: Exception (28) - test sketch

luc-github, 2015-10-01:
Now heap corruption is solved (thank @iggr) here another issue I meet
Using staging : 1.6.5-1160-gef26c5f - I had hard time to do it but I was able to reproduce a random exception(28) out of my project-  here the test sketch:

Just need doing some navigation between push button and page 2 then back to home - no need a big stress and it will happen when display home or page 2
I was able to reproduce more easily using PROGMEM  

I enclose the output and stack for 2 crashes:

First crash you can see memory is still high when crashing
Second crash you can see memory decreasing  during several seconds then crash

<h3>Crash 1</h3>
```
command
31888
refresh
31440
command
31664

Exception (28):
epc1=0x4000df60 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffec570 end: 3ffec890 offset: 01a0

>>>stack>>>
3ffec710:  00002707 00000001 3ffec77c 40209bba  
3ffec720:  00002729 00000001 3ffec77c 40209f85  
3ffec730:  3ffec750 00000074 3ffec77c 3ffeacac  
3ffec740:  3ffeb478 3fff3f08 3ffec77c 40209fe5  
3ffec750:  3ffe9568 3fff3f08 00000001 40209b15  
3ffec760:  0000000a 3ffec840 00000000 40202345  
3ffec770:  3fff40a8 0000003f 00000031 00000000  
3ffec780:  000026ff 000026f8 3fff4088 0000000f  
3ffec790:  0000000d 00000001 3ffec7e0 40209bba  
3ffec7a0:  3fff4068 0000000f 3fff3f08 402085ee  
3ffec7b0:  3ffeb478 3fff3f08 3ffec7e0 40205f7a  
3ffec7c0:  0000000f 3fff42ac 3fff3f08 40205faa  
3ffec7d0:  3fff4160 0000000f 3ffeac78 40205c18  
3ffec7e0:  3fff4068 0000000f 00000001 0000000f  
3ffec7f0:  00000001 3fff40b0 0000000f 00000003  
3ffec800:  3fff46c0 0000007f 00000000 3ffeb7c8  
3ffec810:  40205f84 00000001 402017ec 0000000f  
3ffec820:  402010ae 00000001 00000001 3ffec8bc  
3ffec830:  3fffdc20 000003bd 3ffeac78 40205d2d  
3ffec840:  3ffe9798 00000000 000003e8 000116b2  
3ffec850:  3ffeac90 3fff45f8 00000016 40101a2d  
3ffec860:  4020180d 00000000 3ffec8b4 40202718  
3ffec870:  3fffdc20 00000000 3ffec8b4 4020186a  
3ffec880:  00000000 00000000 3ffeb870 40100398  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```

<h3>Crash 2</h3>
here can see the memory drop

```
refresh
32184
refresh
32184
refresh
32184
refresh
32184
command
21040
refresh
21528
refresh
11728
refresh
11728
refresh
11728

Exception (28):
epc1=0x4000df60 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffec560 end: 3ffec890 offset: 01a0

>>>stack>>>
3ffec700:  00000b43 00000001 3ffec778 40209bba  
3ffec710:  00000b65 00000001 3ffec778 40209f85  
3ffec720:  3f003332 3ffeacac 3ffec778 00000003  
3ffec730:  00000003 40248f70 3ffec778 40209fe5  
3ffec740:  3ffe9568 40248f70 00000001 40209b15  
3ffec750:  3ffeb478 3ffeb53c 00000000 40202545  
3ffec760:  3fff40a8 0000003f 00000031 3fff40a8  
3ffec770:  0000000f 00000002 00000000 00000b3f  
3ffec780:  00000b34 3fff4088 0000000f 0000000d  
3ffec790:  3fff39d0 00000006 3ffec7e0 3ffeacac  
3ffec7a0:  3ffeb478 3fff3f60 3fff3f60 402085ee  
3ffec7b0:  3ffeb478 3fff3f60 3ffec7e0 40205f7a  
3ffec7c0:  0000000f 3fffdc20 3fff3f60 40205faa  
3ffec7d0:  3fff42e0 0000000f 3ffeac78 40205c18  
3ffec7e0:  3fff4068 0000000f 00000006 0000000f  
3ffec7f0:  00000006 3fff4090 0000000f 00000003  
3ffec800:  3fff47c8 0000007f 00000000 3ffeb7c8  
3ffec810:  40205f84 00000001 402017ec 0000000f  
3ffec820:  402010ae 00000001 00000001 3ffec8bc  
3ffec830:  3fffdc20 000003e8 3ffeac78 40205d2d  
3ffec840:  3ffe9798 00000000 000003e8 0003486b  
3ffec850:  3ffeac90 3fff4308 00000016 40101a2d  
3ffec860:  4020180d 00000000 3ffec8b4 40202718  
3ffec870:  3fffdc20 00000000 3ffec8b4 4020186a  
3ffec880:  00000000 00000000 3ffeb870 40100398  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```
here the script

<h3>Test sketch</h3>
```
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
extern "C" {
#include "user_interface.h"
}

const char FILL_THE_VOID1[] PROGMEM ="Fill the page with something visible\n";
const char FILL_THE_VOID2[] PROGMEM ="<Div text='Fill the page with something hidden'>\n";

const char* ssid = "dlink";
const char* password = "1F16264C4D";

ESP8266WebServer server(80);

void handleRoot() {
  String IP = WiFi.localIP().toString();
  String  s;
  s = "<!DOCTYPE HTML>\r\n<html>\r\n<body>";
   s += "<a href='http://"+IP+"'>Home</a>&nbsp;&nbsp;<a href='http://"+IP+"/page2'>Page 2</a>" ;
   s += "<H4>Home</H4>";
   s += "<IFRAME width=\"2\" height=\"2\" style=\"visibility:hidden\" ID=\"frmcmd\" NAME=\"frmcmd\" ></IFRAME><BR>";
   s += "<IFRAME ID=\"statusfrm\" NAME=\"statusfrm\" src=\"http://"+IP+"/STATUS\"></IFRAME>";
   s += "<BUTTON TYPE=\"BUTTON\" VALUE=\"Emergency Stop\" Onclick=\"window.open('http://"+IP;
   s += "/CMD?cmd=M112','frmcmd');\">Emergency Stop</BUTTON><BR>";
   s += "<SCRIPT TYPE=\"text/javascript\">\r\n";
   s += "setInterval(function(){";
   s += "var ifrm=document.getElementById(\"statusfrm\");var doc=ifrm.contentDocument?ifrm.contentDocument:ifrm.contentWindow.document;";
   s += "doc.location.reload(true);},3000);\r\n</SCRIPT>\r\n";
  for (int i=0;i<10;i++)
	{
	s += String(FILL_THE_VOID1);
	}
for (int i=0;i<190;i++)
	{
	s += String(FILL_THE_VOID2);
	}
   s += "</body></html>\n";
  server.send(200, "text/html", s);

}

void handlepage2() {
  String IP = WiFi.localIP().toString();
  String  s;
  s = "<!DOCTYPE HTML>\r\n<html>\r\n<body>";
   s += "<a href='http://"+IP+"'>Home</a>&nbsp;&nbsp;<a href='http://"+IP+"/page2'>Page 2</a><BR>" ;
   s += "<H4>Page 2</H4>";
  int n = WiFi.scanNetworks();
  if (n == 0)
    s += "<H3>no networks found</H3>";
  else
  {
    s += "<H3>Networks found</H3>";
    s += "<TABLE>";
    for (int j = 0;j < n; ++j)
    {
	   s += "<TR><TD>#";
      // Print SSID and RSSI for each network found
      s += String(j+ 1) + "</TD><TD>";
      s += String(WiFi.SSID(j)) + "</TD><TD>" ;
      s +=String(100+WiFi.RSSI(j)) + "%</TD><TD>";
      if (WiFi.encryptionType(j) == ENC_TYPE_NONE)s+="Not protected";
      else s+="Protected";
      s += "</TD></TR>";
    }
    s += "</TABLE><BR>";
  }
  for (int i=0;i<10;i++)
	{
	s += String(FILL_THE_VOID1);
	}
	for (int i=0;i<190;i++)
	{
	s += String(FILL_THE_VOID2);
	}
   s += "</body></html>\n";
  server.send(200, "text/html", s);

}

void handleStatus() {

  String  s = "<!DOCTYPE HTML>\r\n<html>\r\n<body>";
   s += "Ok";
   s += "</body></html>\n";
  server.send(200, "text/html", s);
  Serial.println("refresh");
  Serial.println(system_get_free_heap_size());
}

void handleCMD() {

  Serial.println("command");
  Serial.println(system_get_free_heap_size());
}

void setup(void){
 Serial.begin(115200);
  delay(10);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  server.on("/", handleRoot);
  server.on("/page2", handlepage2);
  server.on("/CMD", handleCMD);
  server.on("/STATUS", handleStatus);

  server.begin();
}

void loop(void){
  server.handleClient();

} 

```

pgollor, 2015-10-01:
Please read the information from [reference.md](http://arduino.esp8266.com/versions/1.6.5-1160-gef26c5f/doc/reference.html#progmem).

You have to use
```c++
s += String(FPSTR(FILL_THE_VOID1));
```

luc-github, 2015-10-01:
yes I forget to add it in test sketch you are correct, I do it already in my project, but adding it do no solve the issue :
```
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
extern "C" {
#include "user_interface.h"
}

const char FILL_THE_VOID1[] PROGMEM ="Fill the page with something visible\n";
const char FILL_THE_VOID2[] PROGMEM ="<Div text='Fill the page with something hidden'>\n";

const char* ssid = "dlink";
const char* password = "1F16264C4D";

ESP8266WebServer server(80);

void handleRoot() {
  String IP = WiFi.localIP().toString();
  String  s;
  s = "<!DOCTYPE HTML>\r\n<html>\r\n<body>";
   s += "<a href='http://"+IP+"'>Home</a>&nbsp;&nbsp;<a href='http://"+IP+"/page2'>Page 2</a>" ;
   s += "<H4>Home</H4>";
   s += "<IFRAME width=\"2\" height=\"2\" style=\"visibility:hidden\" ID=\"frmcmd\" NAME=\"frmcmd\" ></IFRAME><BR>";
 s += "<IFRAME ID=\"statusfrm\" NAME=\"statusfrm\" src=\"http://"+IP+"/STATUS\"></IFRAME>";
   s += "<BUTTON TYPE=\"BUTTON\" VALUE=\"Emergency Stop\" Onclick=\"window.open('http://"+IP;
   s += "/CMD?cmd=M112','frmcmd');\">Emergency Stop</BUTTON><BR>";
   s += "<SCRIPT TYPE=\"text/javascript\">\r\n";
   s += "setInterval(function(){";
   s += "var ifrm=document.getElementById(\"statusfrm\");var doc=ifrm.contentDocument?ifrm.contentDocument:ifrm.contentWindow.document;";
   s += "doc.location.reload(true);},3000);\r\n</SCRIPT>\r\n";
  for (int i=0;i<10;i++)
	{
	s += String(FPSTR(FILL_THE_VOID1));
	}
for (int i=0;i<190;i++)
	{
	s += String(FPSTR(FILL_THE_VOID2));
	}
   s += "</body></html>\n";
  server.send(200, "text/html", s);

}

void handlepage2() {
  String IP = WiFi.localIP().toString();
  String  s;
  s = "<!DOCTYPE HTML>\r\n<html>\r\n<body>";
   s += "<a href='http://"+IP+"'>Home</a>&nbsp;&nbsp;<a href='http://"+IP+"/page2'>Page 2</a><BR>" ;
   s += "<H4>Page 2</H4>";
  int n = WiFi.scanNetworks();
  if (n == 0)
    s += "<H3>no networks found</H3>";
  else
  {
    s += "<H3>Networks found</H3>";
    s += "<TABLE>";
    for (int j = 0;j < n; ++j)
    {
	   s += "<TR><TD>#";
      // Print SSID and RSSI for each network found
      s += String(j+ 1) + "</TD><TD>";
      s += String(WiFi.SSID(j)) + "</TD><TD>" ;
      s +=String(100+WiFi.RSSI(j)) + "%</TD><TD>";
      if (WiFi.encryptionType(j) == ENC_TYPE_NONE)s+="Not protected";
      else s+="Protected";
      s += "</TD></TR>";
    }
    s += "</TABLE><BR>";
  }
  for (int i=0;i<10;i++)
	{
	s += String(FPSTR(FILL_THE_VOID1));
	}
	for (int i=0;i<190;i++)
	{
	s += String(FPSTR(FILL_THE_VOID2));
	}
   s += "</body></html>\n";
  server.send(200, "text/html", s);

}

void handleStatus() {

  String  s = "<!DOCTYPE HTML>\r\n<html>\r\n<body>";
   s += "Ok";
   s += "</body></html>\n";
  server.send(200, "text/html", s);
  Serial.println("refresh");
  Serial.println(system_get_free_heap_size());
}

void handleCMD() {

  Serial.println("command");
  Serial.println(system_get_free_heap_size());
}

void setup(void){
 Serial.begin(115200);
  delay(10);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  server.on("/", handleRoot);
  server.on("/page2", handlepage2);
  server.on("/CMD", handleCMD);
  server.on("/STATUS", handleStatus);

  server.begin();
}

void loop(void){
  server.handleClient();

} 

```

```
command
32072
command
32072
command
32072
refresh
31400

Exception (28):
epc1=0x4000df60 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffec560 end: 3ffec890 offset: 01a0

>>>stack>>>
3ffec700:  00000c7a 00000001 3ffec778 40209bba  
3ffec710:  00000c9c 00000001 3ffec778 40209f85  
3ffec720:  3f003432 3ffeacac 3ffec778 00000004  
3ffec730:  00000004 40248f70 3ffec778 40209fe5  
3ffec740:  3ffe9568 40248f70 00000001 40209b15  
3ffec750:  3ffeb478 3ffeb53c 00000000 40202545  
3ffec760:  3fff4598 0000003f 00000031 3fff40a8  
3ffec770:  0000000f 00000002 00000000 00000c6f  
3ffec780:  00000c6b 3fff4088 0000000f 0000000d  
3ffec790:  3fff39d0 00000006 3ffec7e0 3ffeacac  
3ffec7a0:  3ffeb478 3fff3f60 3fff3f60 402085ee  
3ffec7b0:  3ffeb478 3fff3f60 3ffec7e0 40205f7a  
3ffec7c0:  0000000f 000007d1 3fff3f60 40205faa  
3ffec7d0:  3fff4290 0000000f 3ffeac78 40205c18  
3ffec7e0:  3fff4068 0000000f 00000006 0000000f  
3ffec7f0:  00000006 3fff4068 0000000f 00000003  
3ffec800:  3fff46f0 0000007f 00000000 40204ba9  
3ffec810:  40205f84 00000001 000003e8 0000000f  
3ffec820:  00000000 3fff46b8 3ffeac78 3ffec8bc  
3ffec830:  3fffdc20 000003e9 3ffeac78 40205d2d  
3ffec840:  3ffe9798 00000000 000003e8 00016dd8  
3ffec850:  3ffeac90 3fff46b8 00000016 40101a2d  
3ffec860:  4020180d 00000000 3ffec8b4 40202718  
3ffec870:  3fffdc20 00000000 3ffec8b4 4020186a  
3ffec880:  00000000 00000000 3ffeb870 40100398  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

```

pgollor, 2015-10-01:
What do you want to simulate with your for loops?
Without the for loops it works fine.

luc-github, 2015-10-01:
the loop it is just to fill the page with content like my html pages use lot of content from progmem to save space.
The sketch emulate all actions done in my project in simple sketch to reproduce issue, the loop is the easy way to generate page with same size as it could be one of the factor to reproduce issue

luc-github, 2015-10-01:
Is there a reason loops could be the root cause ?

pgollor, 2015-10-01:
No that was a general question from me.

I am not sure but i think it is not the best way to load so much data from flash into a String class. It should be better to put the data directly from flash to the tcl client. But it is not so easy to handle with dynamic data.

luc-github, 2015-10-02:
I agreed, now SPIFFS is working well, I plan to use it with a template engine instead of lot of PROGMEM, but I will still use some PROGMEM, so issue will still be there.
This sketch is just an easy way to see the issue and hope to find a way to solve it.

luc-github, 2015-10-02:
I do not know if it is helping but I just received a NodeMcu Lua v2, I guess it is 1.0 as it is black.
If I flash above test sketch to it with FPSTR on PROGMEM strings , it crash at first launch of page. but  if I use strings only and not PROGMEM it is ok, so it is more sensitive than my ESP01 and may confirm issue in usage of PROGMEM.

igrr, 2015-10-02:
Initial analysis shows that the error happens when we hit a limit of available heap memory while doing string manipulations. This becomes quite obvious with `Serial.setDebugOutput(true);` set, because there are some debugging messages from the mem_manager.o module about failed allocations.
I think in this case we should halt the execution and print a decent error message.

luc-github, 2015-10-02:
Thanks a lot @igrr, but for my information why this not happening if not using PROGMEM ? does usage of FPSTR need extra memory ?

luc-github, 2015-10-04:
my question is to understand which size of string, passed to send function,  is critical and which size can be considered safe compare to available memory.

as I plan to build pages part by part, by using sendcontent for each part when completed, then build the next part.

I plan to build/send 2048 sized parts but would be sure it is safe size when using PROGMEM strings or SPIFFS files used line by line to replace variable elements

luc-github, 2015-10-16:
I have limited the page code generation by using html template on SPIFFS, limiting the usage of String object to minimum and build/send 1.2K~1.4K sized parts at once, so currently on the 23K of available memory I only use 4K and do not have anymore memory availability issue due to big page creation.

Thanks

closed 2015-10-16 by luc-github

pgollor, 2015-10-16:
Can you post a sample code with your solution? It would be great.

luc-github, 2015-10-16:
Sample code won't be easy as it depend how you implement it 
you can have a look to my project : https://github.com/luc-github/ESP8266 - and especially webinterface.cpp (https://github.com/luc-github/ESP8266/blob/master/esp8266/webinterface.cpp), I still need to rewrite the readme.md with more explanation how to use the keywords but the concept is the following: separate UI from dynamic data, handle small packets in memory and avoid Strings as much as possible.

The template files (https://github.com/luc-github/ESP8266/tree/master/esp8266/data) use html file with variable parts replaced by keywords (content of another file, information system, dynamic list, etc...) here the list https://github.com/luc-github/ESP8266/blob/master/keywords.txt

I stored the keys in PROGMEM and generate the values according the needs - to store the KEYS that need to be replaced by the VALUES - I use a linked list of array of char to limit the memory usage and be sure memory is released when I decide, not randomly like with String - https://github.com/luc-github/ESP8266/blob/master/esp8266/storestrings.cpp

then I give the KEYS / VALUES list and template file name to a function (https://github.com/luc-github/ESP8266/blob/master/esp8266/webinterface.cpp#L269) which first calculate the final page size to create the correct content-length for the header (https://github.com/luc-github/ESP8266/blob/master/esp8266/webinterface.cpp#L415) then build and send page itself part by part but which size under 1.4K.

To limit the memory usage, the function actually just read template file and include files, line by line (https://github.com/luc-github/ESP8266/blob/master/esp8266/webinterface.cpp#L301), replacing KEYS by VALUES when found (https://github.com/luc-github/ESP8266/blob/master/esp8266/webinterface.cpp#L303) - and put read/processed line in a buffer  - when buffer size is over 1.2K,(https://github.com/luc-github/ESP8266/blob/master/esp8266/webinterface.cpp#L413) send it and reset buffer.
Nothing clever just a better control of building page process based of issues I meet - my code is far from being perfect but so far not more memory issues

I hope the explanations helped

pgollor, 2015-10-16:
Thank you for your last post. That's more then expected.
I will look at your repository.