esp8266/Arduino #845: Old bug is back - NodeMCU 0.9 upload error

mozgy, 2015-10-01:
Updated to latest staging and old bug is back ->

warning: espcomm_sync failed
error: espcomm_open failed

keeping FLASH button pressed upload from IDE works ..

myrtleTree33, 2015-10-03:
+1 for this.

thinkl33t, 2015-10-14:
Can confirm this, reverting back to stable has allowed me to upload again.

seclorum, 2015-11-15:
The problem is plainly-wrong reset sequence involving DTR .. I fixed it in my local esptool.py .. but could not find it in esp8266/Arduino (latest), (alas it seems esptool is a binary now, wtf?)  .. anyway, my fix:

diff --git a/esptool.py b/esptool.py
index b76f120..c5459c7 100755
--- a/esptool.py
+++ b/esptool.py
@@ -138,7 +138,7 @@ class ESPROM:
             # issue reset-to-bootloader:
             # RTS = either CH_PD or nRESET (both active low = chip in reset)
             # DTR = GPIO0 (active low = boot to flasher)
-            self._port.setDTR(False)
+            self._port.setDTR(True)
             self._port.setRTS(True)
             time.sleep(0.05)
             self._port.setDTR(True)

Links2004, 2015-11-15:
there are two esptools one is based on python the other in C (the C version is included in the IDE)
- esptool.py   https://github.com/Links2004/esptool
- esptool.exe https://github.com/igrr/esptool-ck

thinkl33t, 2016-02-18:
This bug is still present as of 2.1-rc2.  Its a pain in the ass having to have the back of my devices open to reflash them :(

devyte, 2017-10-20:
I can't reproduce this with latest git on a Nodemcu v3 Lolin, flashing seems to work fine.
Closing due to can't reproduce and age.
If you think there is still a problem, please open a new issue, and fill in the requested info.

closed 2017-10-20 by devyte