esp8266/Arduino #849: Canonical bootstrapping approaches to user setting SSID / PSK / Method in deployment?

vicatcu, 2015-10-01:
The same problem every Wi-Fi connected thing ever has to solve... how to get the target network credentials into the device without compiling them into the sketch. 

I've been trying to create a basic example that does this:
 1. Tries to connect to the SDK stored network configuration.
 2. If it can't connect to the SDK stored network configuration (or if the user pushes a button), reverts to AP_STA mode so that you can connect to the ESP8266 with a computer.
 3. In AP_STA mode, runs MDNS and Webserver with a page that you can go to http://esp8266.local in a browser, be presented with a form to fill out the network credentials, and click submit.
 4. Also a form submission handler page that tries to connect the STA to the network and credentials provided by the user and provides success / failure feedback to the user regarding ability to connect. On successful connection this becomes the SDK network configuration.

I've been trying to put something like this together from the examples, and it's proving a bit harder than I thought it would be.  I've managed to kind of do aspects of the above by stitching together code from the library examples, but the WebServer is not very snappy to respond  in the webpage, and I have a heck of a time connecting to the SoftAP (both with my Android phone and my Ubuntu 14.04 laptop, both indicating very strong signal strength). Happy to share my (currently crappy) code [here](http://pastie.org/10454905).

I recognize that this is not *really* an issue per se, an @igrr if you prefer to move this thread elsewhere please chime in.

vicatcu, 2015-10-01:
By the way, the web server is very snappy if I hardcode connect it to my network and access it via the STA IP address...

drmpf, 2015-10-01:
Hi Victor,
Try pfodWifiConfig library and example code detailed on 
http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html
Does most of what you want, but access to the config web server is via 10.1.1.1  
I did not bother to add the name resolution stuff, but happy to ride on the back of your efforts when you get that running.

That example also covers how to generate the AP secure password and write it out as a QR code attach to the ESP device to be scanned to get the config AP password.

I had another user recently request a small Android app that opens a scanner to scan the password and then connects the the AP and open the config webpage, seems to work for him. 

As an aside I also have a process and libraries for configuring WiFi devices that cannot be provide an AP (see http://www.forward.com.au/pfod/pfodWifiConfig/index.html)

pgollor, 2015-10-02:
Did you try wifi WIFI_AP_STA mode?
And try to wait for connection after begin.

vicatcu, 2015-10-02:
@drmpf Thanks, I will definitely have a look!

@pgollor from what I can tell the ESP8266 comes up in AP_STA mode by default. Where are you suggesting I should be waiting for connection in [the code[(http://pastie.org/10454905) that I posted? Bear in mind that before I give the ESP8266 credentials for the SDK to internalize, the STA can't possibly connect to anything, right?

igrr, 2015-10-02:
Yes i think it would be good to do this check inside WiFi.begin.
I am planning to do some refactoring in ESP8266WiFiClass, because the code is getting rather convoluted. Especially the logic for switching between WiFi modes... I will create a separate ticket for that.

Regarding the original topic, yes, it makes sense to provide a ready-made solution for initial configuration, probably in the form of a library. Some aspects about dependencies we may have in such a library are not clear to me, like:
- is it okay to require the user to include ESP8266WebServer library?
- if we use the ESP8266WebServer library, who gets to initialize the server? Should it be done inside the library or the user has to do it in the sketch?
- can we depend on the file system, or should we use simplified html pages and store them in PROGMEM?
- is it okay to require the user to upload some html files to the file system?

These choices mostly depend on usage scenarios we want to cover. Any discussion which may help clarify these points is welcome!

pgollor, 2015-10-02:
@igrr I think with a stable upload tool for the filesystem it is better to use it by default instead oft progmem pages.

vicatcu, 2015-10-02:
@igrr I think you're thinking about the right things. I'm good with the idea of the capability being encapsulated in a library, so lets explore that idea.

 * Since the user may want to include a Server in their sketch in a more general sense than for initial configuration, I think it makes sense for the Server to be passed to class from the sketch somehow (e.g. a setServer() method).
 * Following the same logic from the previous bullet point, I think it should probably be left to the user sketch to initialize the server. Ideally there would be adequate reflection possible so that the library could complain if it is passed a Server object that is not in the expected state. A solid example with good inline comments about the importance of sequence of events would go a long way.
 * SPIFFS seems really cool and versatile. Uploading one or more files as a pre-requisite for the library seems like a pretty reasonable expectation. Possibly it would be good to have a minimal implementation in PROGMEM if the required files are not found in SPIFFS. I think might be pretty important that an example work "out-of-the-box".

I guess that's my opinions. Interested in others opinions.

vicatcu, 2015-10-02:
@drmpf I was able to try your [pfodWifiConfig sketch](http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/ESP8266_WifiShield_Rev3.ino.txt) but I had a similar experience to my own attempt, i.e. the server didn't respond (reliably/quickly) to the Configure button being pressed, and usually the "/config" page never loads (I've got it to load once). Sometimes I have to connect to the SoftAP a few times, and sometimes it drops me after connecting as well. Could it be just some undiscovered issues in the staging release (server performance on softAP interface), or is your experience similar in general?

vicatcu, 2015-10-02:
Been scouring the web... found [this](http://www.esp8266.com/viewtopic.php?p=17694) forum post that I'll need to give a try. I'll give it a whirl tomorrow and report back.

drmpf, 2015-10-03:
@igrr
 >is it okay to require the user to upload some html files to the file system?

From my point of view NO, May be OK for a commercial board setup, but for novices, I need the simplest process possible.  I.e. load this sketch and run.

>Since the user may want to include a Server in their sketch in a more general sense than for initial 
configuration, I think it makes sense for the Server to be passed to class from the sketch somehow (e.g. a setServer() method).

I don't think this is necessary or desirable.  When starting in config mode none of the user's normal code will be run so the setting up of the server can and should be done completely within the supporting library.  Not that my example code does this but.. if we are going to provide a standard library that is how I think it should run.

@vicatcu,  I use the pfodWifiConfig regularly without any problems at all. I connect from my Android a few feet away.  Always works for me.  I am using package esp8266-1.6.5-947-g39819f0, which is not the latest.  This suggests to me your Wifi network is flaky.  Perhaps you could try changing Channels for the ESP?  I was never sure how you are supposed to set the channel.  You could also try running WireShark on your pc and connect it to your ESP AP and see what is happening.
Had quick look at the link you posted.  pfodWifiConfig has a more complete URL decoder which handles more then just spaces.  Also see following comments.

@igrr and @vicatcu 
As a general point.  Just setting the ssid and password (and security WPA2 etc) is usually not enough.  If the user is going to run the ESP as a server then they at least want to set the port it will listen on and probably the IP address so they can find it.
If the user is going to run the ESP as a client then they will want to set the host and port to connect to and perhaps the username and password.  (Currently working with a user on this type of client config.)  
Setting up to run as a client is not completely straight forward because of the various types of client to server connections, but at least by storing the server host, port, username and password for the client connection from the config then the sketch can just retrieve them and plug them into the appropriate spots.

So for a general config library I think we need to allow for these setting as well.  Have a look at 
http://www.forward.com.au/pfod/pfodWifiConfig/pfodWifiConfig.html
for a definition of what I think needs to be configured and saved for various uses,  Server, Client, ClientLogin

Also since I wrote that I have found you need to have a connection timeout setting. Essential for running as a Server and probably very useful when running as a Client to ensure the WiFi connection is cleaned up properly when there are network errors, drop outs, router power outages etc.
http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html and the supporting library include configuring the connection timeout.

jeffryr, 2015-10-03:
@vicatu I posted the onboarding code in http://www.esp8266.com/viewtopic.php?p=17694.  There is a slightly (by two days) more up to date version at https://github.com/jeffryr/ESP8266-Sketches.  AFAIK it still works but it's been quite a few months and there may be a more elegant solution.

chrisfraser, 2015-10-03:
This is a loosely related problem I have been thinking about for a while. 

When I have a few ESP's I am working on at the same time, to give them all a generic name causes issues. I recently created a whole lot of boards for controlling robot arms and gave them away at a hackathon. The default AP SSID was eeArm on 50 boards and this resulted in people inadvertently setting up each-others boards. This caused huge confusion.

Does anyone know of a predictable way to create a unique SSIDs? Maybe based on MAC address? Would be cool to create, say a six character unique string. The alternative is burning unique defaults to each, but that seems like a lot of work!

drmpf, 2015-10-03:
Yes MAC address would be an good alternative in that situation. 
```
uint8_t* softAPmacAddress(uint8_t* mac);
    String softAPmacAddress(void);
```
looks like what you want.
I do burn each one with a unique password which I also use for pfodSecurity.

grahamehorner, 2015-10-03:
a simple webpage with some security allowing posting the a configuration file into flash memory is the best option; in my opinion; note the SECRUITY is needed SSL/HTTPS server would be great, but is the unit capable? some RSA public/private key encryption using javascript client side encryption to secure login details and configuration data.

grahamehorner, 2015-10-03:
BTW, I'm working on a simple webserver that has a configuration page (username/password) which allows the upload of a config.json file this has a setting WiFi mode and SSID together with a secured password encrypted using a public/private key.

After initial configuration of the WiFi SSID/password the webserver restart the WiFi using the chosen SSID/password and enabled OTA update features.

drmpf, 2015-10-03:
`a simple webpage with some security allowing posting the a configuration file into flash`

I don't think any extra security is needed.  The ESP configuration AP should be protected by a 'secret' key, in my example that key is unique per device.  So only users with access to the secret key can connect to the ESP AP and load the real network ssid and password.
Others cannot eavesdrop on this connection without the 'secret' key.

drmpf, 2015-10-03:
I am ignoring OTA updates and how they may be handled.  While OTA is a great feature, I don't think is it necessary for the general user wanting to get their device setup on the nextwork.  More for developers making code changes.

vicatcu, 2015-10-04:
Reporting back. This code from the forum, crashes, but thanks @jeffryr for the github link, that doesn't crash. I still get aweful performance from the SoftAP though. Here's a ping report:

Also when I go to the server page and punch in my SSID and password (which loads eventually), when I submit the form it never results never loads, eventually resulting in "Error code: ERR_CONNECTION_TIMED_OUT" in the browser. Meanwhile in the serial monitor, sometime after pressing submit, I get:
```
Waiting for Wifi to connect...
66666666663
Great Success!

Soft WDT reset

ctx: cont 
sp: 3ffec630 end: 3ffec900 offset: 01b0

>>>stack>>>
3ffec7e0:  3fff4460 0000000f 0000000b 3fff4398  
3ffec7f0:  0000000f 0000000c 3ffec8b0 4020abaa  
3ffec800:  3fff4098 00000006 3ffec850 3ffeabfc  
3ffec810:  3ffeb3c8 3fff3ff8 3fff3ff8 402095e2  
3ffec820:  3ffeb3c8 3fff3ff8 3ffec850 4020643a  
3ffec830:  0000000f 3fffdc20 3fff3ff8 4020646a  
3ffec840:  3fff4cb0 0000000f 3ffeabc8 402060d8  
3ffec850:  3fff4378 0000000f 00000006 0000002f  
3ffec860:  00000006 3fff4460 0000000f 00000003  
3ffec870:  3fff5240 0000007f 00000000 3ffeb8b4  
3ffec880:  40206444 00000001 402013f4 0000000f  
3ffec890:  402010ae 00000001 00000001 3ffec92c  
3ffec8a0:  3fffdc20 000003e5 3ffeabc8 402061ed  
3ffec8b0:  3ffe96c8 00000000 000003e8 0006a248  
3ffec8c0:  3ffeabe0 3fff4d60 00000016 40101a2d  
3ffec8d0:  40201415 00000000 3ffec924 402028e0  
3ffec8e0:  00000000 00000000 00000000 40201472  
3ffec8f0:  00000000 00000000 3ffeb8e0 40100398  
<<<stack<<<
```
On a reset it does connect to my network, so that's a plus...

drmpf, 2015-10-04:
Clearly something the in the code is not yielding (yield() or delay()).  It is possible that it is a library bug, BUT.. most likely a problem in the uploaded Arduino sketch. 
You need to start cutting out code until it goes way and then put the last line back in to reproduce the error. (but I didn't nee to tell you that)

 I did not have that issue with http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html after adding some yield() calls.

You might also look at https://github.com/jeelabs/esp-link/tree/v1.0.1  for another set of code

jeffryr, 2015-10-05:
The reset is deliberate.  After successfully connecting to the selected wifi network, the code prints a success message (which admittedly could be more descriptive), waits 3 seconds and then calls abort() to reboot into WIFI_STA mode.  Calling ESP.reset() would probably make the serial output appear a bit less ominous.  The reason for the reboot is because at the time I couldn't figure out a way to reliably:

1) Run in both WIFI_AP and WIFI_STA mode
2) Switch from one mode to the other

It works well enough to initialize the two dozen or so various modules I seem to have accumulated over the past year but could definitely use some improvements to make it a little more user friendly.

tzapu, 2015-10-07:
guys, i ve made a lib for exactly this
https://github.com/tzapu/WiFiManager

works good enough for me, when it can t connect it starts up a webserver in ap mode.

drmpf, 2015-10-07:
@tzapu  A couple of points, 
1) I like your addition of DNS,  I assume that means that you can now access the config via any Url,  like www.configmydevice.com  etc.
2) A minor point, with the autoconnect the way you have it, it is difficult (impossible) to change the config if the current connections AP is still running.
3) as I have mentioned before, config usually involves more then just ssid and password.  An extension supporting Static IP, server portNo, client username, password, client host connection would be good.

Informatic, 2015-10-07:
Just my two cents: I think it would be best to prepare some sort of generic key-value storage with appropriate editor web interface (form generator and parser) first, and then integrate it into some autoconnect config/first-run wizard.

tzapu, 2015-10-08:
@drmpf 
1. correct, but with an ios device at least i found that the login/captive portal dialog on connection is pretty much the fastest way to configure it, no need to start a browser, go to some url, etc
2. i did think of this a lot but due to time constraints this was developed on a 'i need to do that' basis, and so far i ve only had to configure nodes once when setting them up and once when putting them into the final location. having a webserver on at all times seems overkill for this (at least to me and my use cases) but as a future enhancement i m thinking to make some function that would start up config mode in station mode on demand (like press of a button, or mqtt message)
3. i considered key value storage for an arbitrary number of settings, like api tokens for various services, but i really did not think of static ip, port, etc. my aim was to make it fast and simple but that won t cater for all scenarios of course.

my next steps with it would be to move from eeprom to fs and after that to allow configurable values to be saved. does anyone see moving to fs as a bad move?

@igrr i think (as i ve also done) that the library should rely on the user for as little as possible and initialise everything, take care of connecting, etc. uploading some files would make for a better interface, but i think it s just too complicated at the current point in time to require of people

guys, any modifications and pull requests, are more than welcome

igrr, 2015-10-08:
In general, :+1: WiFiManager looks like a good starting point for many users. The the code is quite easy to follow so i can see people customizing this library to better suit their use case, if necessary.

@tzapu I would like to change a few things, let me know if you would welcome these changes as pull requests in your repository:
- No need to use EEPROM because SDK libraries maintain SSID and password in flash anyway. We can get stored values from there. Two functions are available in staging: `WiFi.SSID()` and `WiFi.psk()` return current STA settings.
- Pages should use PROGMEM instead of .rodata strings. This is not very well supported in stable, but in staging it seems to work fine.
- Refactor examples so that WiFiManager is used as a local variable in `setup()` method. Once its business is done, there is no need to keep it around (along with WebServer instance).
- WebServer should not be a global, it is only needed while acting as a hotspot (related to the previous point).

tzapu, 2015-10-08:
@igrr that sounds absolutely great and i would love a pull request. to be honest most of what you mentioned goes above my current level of arduino understanding so it will be such a great learning opportunity for me as well :D
i ve just pushed a new update with a bit of cleanup and debug messages enable/disable

tzapu, 2015-10-08:
@igrr i ve read and reread and it makes more and more sense what you said.
because of the proposed changes you made, with no eeprom or fs, the possibility of storing third party key/value pairs will be removed as the sdk only stores ssid and psk.
is there any possibility, for people that want some extended version of it that can have tokens and whatnot to have some conditional includes of libraries and code? (can we check in a lib if another lib has been included in the main app?)
Or maybe a more advanced version should be a fork of this...
This is really more of a general arduino question than related to this so sorry for the offtopic.

igrr, 2015-10-08:
Library dependency tracking is coming in the next IDE release (it is already available in nightly builds), so conditionally including EEPROM/FS is not out of the question. This said however, i think it would be in line with general Arduino philosophy to provide a simple tool first, instead of trying to cover all potential use cases. Let's bundle this library and see what people make of it. After all, Arduino is more about making than having things ready-made :)

I like the idea of generic key-value storage though. Done right, it might even be more useful than a file system. But this is indeed a topic for a separate discussion.

tzapu, 2015-10-09:
@igrr i believe i've implemented the last two points you made and pushed to the repo
the other 2 points i don't know how to sort yet so if you ever find the time for that pull request it would be more than welcome. here s what i didn't know/went wrong:
-when are WiFi.SSID() and psk saved?
-i tried using PROGMEM but am getting errors like " error: section attribute not allowed for" and "error: initializer-string for array of chars is too long". do, in short, don't know what i'm doing.

tim-in-oakton, 2015-10-23:
I took a swing at this in our ESP beer keg - all the credentials (SSID, password, AWS credentials, endpoints, region, SNS ARN) are first configured using a web page served by the ESP8266, then persisted to EEPROM.
Model implentation at https://github.com/tim-in-oakton/ESP8266Kegerator 

I will be refactoring this shortly into a "cloud-modem" package that will be much more flexible (multiple SSID/PW combinations, password protected web server, but the above code as-is allows zero-wire config of network access and credential/endpoint data without ANY compiled values being provided.

HTH, and please comment if you have strong feelings about ways to improve this

drmpf, 2015-10-23:
I suggest you add a password to your config AP otherwise anyone can sniff while you are configuring your real network password and so gain access to your real network.  
This page, http://www.forward.com.au/pfod/secureChallengeResponse/keyGenerator/index.html 
has a key generator that writes out a QR code you can scan to get access to your config AP.
I use this system for my Uart to WiFi bridge (http://www.forward.com.au/pfod/CheapWifiShield/index.html)

tzapu, 2015-12-08:
@igrr 
sorted this as well
> No need to use EEPROM because SDK libraries maintain SSID and password in flash anyway. We can get stored values from there. Two functions are available in staging: WiFi.SSID() and WiFi.psk() return current STA settings.
just the PROGMEM bit left

tzapu, 2015-12-23:
@igrr i _hope_ i sorted all of the stuff on your list related to WiFiManager including some other changes mentioned here. most of the strings are now in PROGMEM, again, I _hope_ properly.
if you ever get a sec, maybe you can take a look and see if there s stuff you d like to see done differently or feel it s missing something.

devyte, 2017-10-20:
The WiFiManager fulfills the requirements listed here, and is now mature. Closing.

closed 2017-10-20 by devyte