esp8266/Arduino #851: GET or POST from esp8266

emilianocasas, 2015-10-02:
Hi guys, I have a question. Do I have to create a socket connection to send info to a url? Or there is a function like get or post that simply does that?
Thanks

asadujjamandm, 2015-10-03:
Hey I have been facing a problem about sending GET and POST request using AT command, I am using Arduino IDE, I am trying to send GET request from serial monitor (Arduino IDE) using AT command. But the problem is I failed to send GET/POST request from serial monitor (of Arduino IDE).

AT+RST

AT+CWJAP="ssid","password"

AT+CIPSTART="TCP","shotlu.usrs0.com",80

AT+CIPSEND=75
>GET /receiver.php?apples=56&oranges=23 HTTP/1.0\r\nHost: shotlu.usrs0.com\r\n\r\n
Those are my command pattern for sending a GET request;

Now the problem is AT+CIPSEND dose not working on Arduino IDE,  When I am sending GET request it shows me this message; 

"AT+CIPSEND=75
> GET /receiver.php?apples=56&oranges=23 HTTP/1.1\r\nHost: shotlu.usrs0.com\r\n\r\nbusy s...

SEND OK

+IPD,168:<html>
<head><title>400 Bad Request</title></head>
<body bgcolor="white">
<center><h1>400 Bad Request</h1></center>
<hr><center>hosting</center>
</body>
</html>

OK
CLOSED

OK"

(By those command I am trying to hit a php file which is called receiver.php;  )

But I have checked those command in different ide which is calle RealTerm. Where I have successfully sent GET request.But Don't know why I am getting those message from Arduino IDE. Please guys help me, if you know something about this problem.

Links2004, 2015-10-03:
@asadujjamandm 
this Project is a Port of the arduino code base the the ESP8266 SoC.
its a complete replacement for the AT firmware. 
to run it you not need an atmega ship all code runes on the ESP chip itself.
see: https://github.com/esp8266/Arduino/blob/esp8266/README.md

asadujjamandm, 2015-10-03:
@Links2004 
sorry I didn't get it, can you please tell me more about it? :o

Links2004, 2015-10-03:
basically you can run any Arduino code direct on the ESP chip itself.

first steps:
https://github.com/esp8266/Arduino/blob/esp8266/README.md#installing-with-boards-manager

supported boards:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/boards.md

asadujjamandm, 2015-10-03:
@Links2004 hey man thnx!!! for your reply, basically I am new at this platform, so I have no clear idea about this. In my project there is a sensor which can take data from environment, now I need to send those data to a web server, for that I saw some video tutorials in youtube where got to know about AT command. Can you please tell me is that possible to send data to a web server using those AT command from serial monitor (of Arduino IDE).

Links2004, 2015-10-03:
no AT commands, no atmega chip, only the ESP + Serial adapter.
https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/using-arduino-ide
http://iot-playground.com/2-uncategorised/38-esp8266-and-arduino-ide-blink-example
https://learn.sparkfun.com/tutorials/esp8266-thing-hookup-guide/installing-the-esp8266-arduino-addon

asadujjamandm, 2015-10-03:
But I am using atmega chip (arduino uno) for the sensor, if i use ESP+serial adapter how can i connect  them?

penfold42, 2015-10-03:
There are GPIOs, SPI and I2C available on the ESP8266.

It is not 5v tolerant. If your sensors can run at 3v3 then great. If they need 5v you will need a level shifter

drmpf, 2015-10-04:
@asadujjamandm   
You might like to look at http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html
which provides a UART to WiFi connection.

Actually as presented there it runs as a Server, where as I think you are looking for a client to connect to a web server and then issue Post and Get messages from your UNO to the serial port (D0,D1) and have the ESP send them via WiFi to the web server.

I am in the process of helping another user set up a client connection, but not finished yet.  

If the above is what you want to do then can I suggest you start by getting 
http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html  up and running and then drop me an email at support via pfod.com.au and we can look at doing the client version.

drmpf, 2015-10-04:
also check out https://github.com/jeelabs/esp-link/tree/v1.0.1

asadujjamandm, 2015-10-04:
@penfold42 @drmpf I have successfully connect  ESP8266 module with my arduino uno, with the help of level shifter and 10microF capacitor. As I told I am new at this platform and I saw some tutorial in youtube, so I taught that I can send my data to a web server through AT command. At this time I am little bit confused about the whole thing, cause I can connect this wifi module with internet via wifi network, and for that I used AT command which is AT+CWJAP="ssid","pasword", and I got "OK" message from the serial monitor. Now the question is why I can not send GET or POST request using AT+CIPSEND ? 

This is my connection for arduino uno to esp8266
[image]

drmpf, 2015-10-04:
I gave up on the 'AT' command set. I wrote a library to support it, but found the async msgs coming from the ESP made to too difficult to get a bullet proof version. So now I program the ESP directly from the Arduino IDE using this excellent board support.

asadujjamandm, 2015-10-04:
@drmpf i have wasted  too much time behind this At command :( . In my project I just wanted to send my data to a web server. I used a sensor in my project and this sensor is connected with arduino uno, as I said I have wasted too much time, so need some easy idea for sending my data to a web server, and at this time I am totally blank, so guys please help me :'( .

Links2004, 2015-10-04:
connect the sensor direct to the ESP use the Arduino IDE for ESP8266 to Program the ESP direct to send the data and you are done.

drmpf, 2015-10-04:
@asadujjamandm  see my suggestions above

closed 2015-10-17 by igrr

AdamMiltonBarker, 2015-12-28:
This is an absolute nightmare!!!! All over the internet you see about how you can use this to add WIFI connectivity to Arduino projects but this is not the case at all. I can get a web server up running on the ESP8266 via the Arduino IDE, visit the ip address of the server and literally 20 minutes later the page will load. You can't use it to send MQTT messages or receive from the Arduino, it won't pull data from a webserver when calling the page from an Arduino, I have just wasted a week with this and now can't complete projects on time as a direct result of this. Why is it made out that this is a simple way to add WIFI to an Arduino that is a lie. If I wanted to build a device with the ESP8266 I wouldn't of used an Arduino, it seems I am not the only one has been tricked into thinking that this is a solution for WIFI on an Arduino project.

drmpf, 2015-12-28:
Works well for me in server mode (client connects to ESP8266 TCP server or Web server for config)
see http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html

AdamMiltonBarker, 2015-12-28:
I have the ESP8266 transceiver module not a shield absolutely none of the examples on line have worked, as far as I can see you cannot use this to simply add wifi to Arduino projects, nor can you use it with MQTT on the arduino you have to mess around setting it up in IDE, program on the chip and not the Arduino, if you can send me to the code that proves me wrong then I would actually be grateful but as it is this has cost me a lot of time and caused a massive amount of problems, I have two days to get this and a number of other projects done, none of which is going to happen thanks to this.

AdamMiltonBarker, 2015-12-28:
Can you send me to the tutorial that shows how to successfully add the module to an existing Arduino project and use it to send and receive MQTT messages and HTTP requests because after one week of going through pretty much every tutorial on the web I have not been able to achieve this.

penfold42, 2015-12-28:
By "existing arduino project" do you mean a separate arduino board with something like an atmel avr chip on it ?

If so, this is not the repository for you.

AdamMiltonBarker, 2015-12-28:
just exactly as I said, using the ESP8622 to add WiFi to a project built on an Arduino Uno

penfold42, 2015-12-28:
This is not the repository for you.

This is a port of the arduino framework to run NATIVELY on the 8266 boards.

AdamMiltonBarker, 2015-12-28:
I already knew that, as was my point, the ESP8266 is advertised as a way to add wifi to existing Arduino projects but it is not.

penfold42, 2015-12-28:
If you want to add a simple wifi interface to an existing arduino avr/arm board project that was designed around a serial interface, it works great.

If you want anything more than a simple TCP connection, you've picked the wrong architecture with your approach. Doing it natively on the 8266 will always be 2-10 times easier.

If you feel you've been misled, complain to espressif or dozens of bloggers making claims you've misunderstood - please don't hijack an old closed issue on a repository that has nothing to do with your chosen architecture.

AdamMiltonBarker, 2015-12-28:
I haven't misunderstood anything I am capable of reading and understanding English, sorry for the high jack but there are people asking the same in this post which is what brought me here. So you say you can add the ESP8266 to an existing Arduino board, this is what I am trying to do so how do you do this ?

penfold42, 2015-12-28:
https://github.com/itead/ITEADLIB_Arduino_WeeESP8266

AdamMiltonBarker, 2015-12-28:
So that library allows you to use the ESP8266 to allow the Arduino to connect to the internet ?

kudosabhay, 2016-05-03:
Guys, I have a question. 
I have my Arduino UNO connected to ESP8266, both are working well.
Whenever I put AT commands from the Arduino Code to the Serial Monitor, they run accurately, except for the '>' command. 
What I do is as follows: 
1. AT+CIPSTART="TCP","api.thingspeak.com",80
2. AT+CIPSEND=40
3. > GET /HTTP/1.1
       api.thingspeak.com/apps/thinghttp/send_request?api_key=VD9G5A50GGKH01J0
4. This replies me with : 
    <!DOCTYPE html>
<html>
<head>
  <title>The page you were looking for doesn't exist (404)</title>
  <style type="text/css">
    body { background-color: #fff; color: #666; text-align: center; font-family: arial, sans-serif; }
    div.dialog {
      width: 25em;
      padding: 0 4em;
      margin: 4em auto 0 auto;
      border: 1px solid #ccc;
      border-right-color: #999;
      border-bottom-color: #999;
    }
    h1 { font-size: 100%; color: #f00; line-height: 1.5em; }
  </style>
</head>

<body>
  
  <div class="dialog">
    <h1>The page you were looking for doesn't exist.</h1>
    <p>You may have mistyped the address or the page may have moved.</p>
  </div>
</body>
</html>

Can you please let me know, why is it always saying : Page Doesn't exist? 
Reply as soon as you could. It's a bit urgent.

igrr, 2016-05-03:
You are requesting `http://api.thingspeak.com/api.thingspeak.com/apps/thinghttp/send_request?api_key=VD9G5A50GGKH01J0`.
Opening this URL in browser shows the same "the page you were looking for doesn't exist" thing, so nothing unexpected here.

kudosabhay, 2016-05-03:
Yep. I figured out that one. Thanks for that. 
And, I have got my Answer to my Question :
After the '>' command, the code I have written in the above mentioned steps is wrong.
It actually must be: 
>  GET /apps/thinghttp/send_request?api_key=689WYA85WW1VJSBA HTTP/1.0
    Host: api.thingspeak.com:80 . 
It replies me with my proper required answer. And, thanks for your help too. 

Check the Proper Answer if you need, here http://arduino.stackexchange.com/questions/23780/how-to-capture-data-from-a-webpage-in-arduino-and-display-it-over-serial-monitor

kluszon, 2016-09-20:
@asadujjamandm 

Did you found any solution for your problem with GET request and response "400 bad response"? I've got the same problem :/

riteshp887, 2017-03-30:
Hey people!
I am also doing the same thing on a local server. I am calling a php script using AT commands. I am doing it right from the serial monitor so, i have not developed a code for it. I am listing out the AT commands. 
AT+CIPSTART="TCP","192.168.43.12",80
AT+CIPSEND=90
GET /pandeyji/post.php?sen1=p&sen2=q&sen3=r HTTP/1.0\r\nHost: 192.168.43.12:80\r\n\r\n\r\n

THIS ONLY REPLIES ME WITH SEND OK. There's no problem with the php script or anything else. it's just something in the format i think. please help me out guys. I am stuck with this. Thank You.

kluszon, 2017-04-01:
@riteshp887 
Hello :) Which firmware version have you got? Are you using Arduino IDE and Serial Monitor to communicate with module?

riteshp887, 2017-04-01:
@kluszon . Yes , i am using arduino ide and the serial monitor to communicate with the module. The firmware version: 0.25.0.0 (June 5 2015 16:27:16) SDK version: 1.1.1 Ai- Thinker technology. Thanks for helping :)

kluszon, 2017-04-01:
Have you got ESP connected to Arduino board? If yes, I don't know exactle, why it don't work in Arduino IDE, but I got the same problem. Probably in Arduino IDE in Serial Monitor you sending command to slow (You can't do this faster insert command by hand). You should write a sketch or connect ESP directly by USB converter to PC (without Arduino board and after that GET and POST request work fine in Arduino Serial Monitor and RealTerm).

riteshp887, 2017-04-01:
@kluszon It works fine with real term. But actually i have to send co-ordinates using gps module. I do need arduino for getting the coordinates. I don't know what to do now. :/

kluszon, 2017-04-01:
From sketch too?

riteshp887, 2017-04-01:
@kluszon I meant to say that if am sending the same commands using serial term with the same setup i.e the arduino connected to esp8266 , i am getting the results. I am able to call the php file then. But when i am working with arduino ide. I am not able to get any results.

kluszon, 2017-04-01:
@riteshp887 Try me solution from link. I got the same problem "SEND OK" or "BAD REQUEST 400" form Arduino IDE without result in database. In my situation it was depend from ESP firmware version and delay in Arduino sketch.