esp8266/Arduino #852: WiFiClientSecure() as stream

mtnbrit, 2015-10-03:
Hi I'm trying to do an OTA update by downloading the bin file over https. The connection is made and the fingerprint is verified just fine, but the Updater errors out with ERROR[5]: Stream Read Timeout.

Can a WiFiClientSecure socket be used as a stream currently?

igrr, 2015-10-06:
You should be able to do that, but perhaps there is some bug there. I'll take a look.

igrr, 2015-10-07:
Probably the same issue as reported near the end of #43.

igrr, 2015-10-17:
The issue is probably due to the fact that your HTTPS server uses large TLS record size, which doesn't fit into memory available on the ESP8266. Sadly, it seems that [nginx doesn't have an option](https://insouciant.org/tech/ssl-record-sizes-in-the-wild/) to set TLS record size, as does Apache.
I have implemented [a TLS extenstion](https://tools.ietf.org/html/rfc3546#section-3.2) which allows for record size negotiation, but it doesn't seem to be supported by many servers.

igrr, 2015-11-30:
WiFiClientSecure supports full 16k fragments in the latest release.

closed 2015-11-30 by igrr