esp8266/Arduino #855: Un-Borking a module?

vicatcu, 2015-10-04:
Hey guys,

I'm not sure what happened, but whenever I upload a sketch now, even though it says "Done Uploading" without any errors,  when I put it into back into Flash bootmode and reset the unit, all it does is spew garbage out the Serial port. I've tried every possible baud rate in the IDE, but there's no baud rate that results in legible English ASCII output. Looking at the output with a scope I'm seeing what appears to be a 13.34us bit time, which would be like 75kbaud.

The last thing I did before I noticed this happening was load this sketch:

```void setup() {
  // put your setup code here, to run once:
  delay(1000);
  Serial.begin(115200);
  Serial.print(F("Flash Chip Size By Id: "));
  Serial.println(ESP.getFlashChipId());
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

And the one time I got an output before that using getFlashChipSizeByChipId it returned printed 0.  Anyone seen anything like this happen before? How to recover?

vicatcu, 2015-10-04:
So, I went over to Windows and used flash_download_tool_v1.2_150512 from the SDK and loaded:
 * AT_bin_v0.25_for_512KB_flash_cannot_upgrade\eagle.flash.bin @ 0x00000 
 * AT_bin_v0.25_for_512KB_flash_cannot_upgrade\eagle.irom0text.bin @ 0x40000

I'm using an ESP-WROOM-02module. The Flash Download tool outputs this stuff:
```
flash vendor:
A1h : FM
flash devID:
4016h
QUAD;32Mbit
crystal:
26 Mhz
```
After loading that stuff, it still spews junk to the console, but it's actually something like 115200 baud and if I scan the output (most of it is still nonsense text), I see "C MEM CHECK FAIL!!!"  being output occasionally.

vicatcu, 2015-10-04:
Seems pretty similar to https://github.com/esp8266/Arduino/issues/95

vicatcu, 2015-10-04:
Yup 77400  output read:

load 0x4010f000, len 1264, room 16
tail 0
chksum 0x42
csum 0x42
~ld

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

... over and over again ...

vicatcu, 2015-10-04:
Based on issue #95, I grabbed the latest esptool from [here](https://github.com/themadinventor/esptool) and tried running:
```
python esptool.py --port COM3 --baud 76800 erase_flash
```

Right after doing that, and switching the boot mode back to Flash and brought up a PuTTY console at 77400, and got this output:
```
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

ets_main.c
```

This would seem to imply that my boot mode is incorrectly set? So looking at the GPIO0 pin with a scope, what I see is something like a sine wave at 26MHz, min 0.94mV, max 2.98V. It's pulled up to 3.3V through a 10k pull-up, is that too weak?

vicatcu, 2015-10-04:
So I tried connecting GPIO0 to 3.3V directly, and then to GND directly, this is right after doing the flash erase with esptool.py. 

The output when I connect GPIO0 to 3.3V is:
```
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

ets_main.c
```

The output when I connect GPIO0 to GND is: 
```
 ets Jan  8 2013,rst cause:1, boot mode:(1,7)
```

Then loading a trivial sketch using the Arduino IDE...

The output when I connect GPIO0 to 3.3V is (repeatedly):
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16
tail 0
chksum 0x42
csum 0x42
~ld
```

The output when I connect GPIO0 to GND is (still): 
```
 ets Jan  8 2013,rst cause:1, boot mode:(1,7)
```

vicatcu, 2015-10-04:
.... and I'm out of ideas ... :8ball:

Links2004, 2015-10-05:
I have seen simulate problems with a damaged flash. 
can you dump the SPI flash with external hardware and compare it to the bin file you download?

vicatcu, 2015-10-05:
Downloaded esptool on Ubuntu, appears to be something very messed up (no surprise here). 
```
python esptool.py --port /dev/ttyUSB0 flash_id
Connecting...
Manufacturer: a1
Device: a1a1
```
Suspicious 'a1' everywhere... then did:

```
./esptool.py dump_mem 0x40000000 65536 iram0.bin
```

... and using some Bless Hex export magic, that bin has the contents [here](https://gist.github.com/vicatcu/7687ea746ced6e9843f2). Now that means next to nothing to me, but maybe someone else can glean insight from that?

Links2004, 2015-10-06:
you have dumped stuff from the RAM.
for flash use:
```
./esptool.py read_flash <address> <size> <outFile>
```
when the flash is OK you will get the exact same binary like the compiler will build. 
the read_flash is not fast it will take some time. if the compare is not the same you flash chip is defective and need to be re-pleased.  

it goes a lot faster it you have access to something like a ```CH341A USB Programmer```,
in combination with a ```SOIC clip``` you not need so solder.

vicatcu, 2015-10-06:
@Links2004 Yea I did the read_flash and compared it to the bin generated by the build process and they matched. I think we probably should just chalk this up to "mysterious ESD goblins" and close the issue. If it comes up again we can re-open it. If I'm the only one who has seen this happen, it's not a real issue.

closed 2015-10-06 by vicatcu

Cieszlunek, 2018-01-09:
Hi there. I know I'm late wit answer, but maybe I'll help someone else who is looking for answer:
Issue is related to voltages missmatch on line ESP RX -> UART TX. ESP RX accepts 3.3V, but UART produces 5V. So it's needed some voltage reduction, i.e. by using two resistors R1 = 4.7k and R2 = 10k.
Connection: ESP RX -> R2 (connected to mass) -> R1 -> UART TX
[image]

mharmon12, 2018-04-15:
I got this.
so when you see,
ets Jan  8 2013,rst cause:1, boot mode:(1,7)

That means you are in program mode.

Remember, the GPIOs are internally tied high on ESP-01(s)
To get there:
GPIO-2 (RST=Reset) to GND
GPIO-0 to GND (Program/Flash mode pin)
Reset off
GPIO-0 off

So you are ready to flash!  :)

After flash remove the jumper from GPIO-0
Then reset by cycling power or touching Reset to GND momentarily.

Now when you reset then baud rate is 74880 odd value.
You'll see the memory status and a little garbage at the end.

Setting buad rate to 115200 and then resetting will now print that same memory dump (it will look like junk) then Ready at the bottom.

Send AT to put into AT mode
and then send AT+GMR to see the current AT version and such.

You will not see any AT messages or ready IF you flash over it with your own binary or Arduino Sketch.

One last note, you NEED to make sure to have a robust 3.3 or if you can adjust to 3.6 BUT that is only if you are having issues flashing.

Good luck and have fun!
MIKE <OUT>