esp8266/Arduino #861: Call function with arguments eat 90 bytes of code? Why?

robertgregor, 2015-10-06:
Hello, I have the following code:

String buildSubmitWithValue(const String& name, const String& value) {
  String ret = F("<input type='submit' name='");
  ret += name;
  ret += F("' value='");
  ret += value;
  ret += F("'/>&nbsp;&nbsp;");
  return ret;
}

Then I will call it:
form += buildSubmitWithValue(F("pot"), F("Power on with timeout"));

The problem is, that each call of the function will eat around 90 bytes of code, which is I think quite a lot. So in my case, it doesn't almost make sense to do it, thus if I will put the full string like this, it will occupy less bytes:
form += F("<input type='submit' name='pon' value='Power on'/>&nbsp;&nbsp;");
form += F("<input type='submit' name='pof' value='Power off'/>&nbsp;&nbsp;"); 

I wanted to reduce the code space, with this, but at the end, it occupy more code space :)

Thanks everyone for the help.

igrr, 2015-10-07:
Here's the test sketch:

```c++
String buildSubmitWithValue(const String& name, const String& value) {
  String ret = F("  ");
  return ret;
}

String test() {
  String form;
  form += buildSubmitWithValue(F("pot"), F("Power on with timeout"));
  return form;
}

void setup() {
  Serial.begin(115200);
  Serial.println(test());
}

void loop() {}
```

Here's the code generated for `test()` function which we are interested in. I've added some comments:
```asm
40201fdc <test()>:
40201fdc:	fffc31        	l32r	a3, 40201fcc ; load empty string ("")
40201fdf:	c0c112        	addi	a1, a1, -64 ; function prologue: adjust stack
40201fe2:	f109      		s32i.n	a0, a1, 60 ; save registers
40201fe4:	e1c9      		s32i.n	a12, a1, 56
40201fe6:	02cd      		mov.n	a12, a2 ; this will later be the return value
40201fe8:	011905        	call0	4020317c <String::String(char const*)> ; call constructor for String form
40201feb:	fff931        	l32r	a3, 40201fd0 ; load "pot"
40201fee:	18c122        	addi	a2, a1, 24 
40201ff1:	014c45        	call0	402034b8 <String::String(__FlashStringHelper const*)> ; call constructor for argument 1
40201ff4:	fff831        	l32r	a3, 40201fd4 ; load "Power on with timeout"
40201ff7:	21cb      		addi.n	a2, a1, 12
40201ff9:	014bc5        	call0	402034b8 <String::String(__FlashStringHelper const*)> ; call constructor for argument 2
40201ffc:	fff731        	l32r	a3, 40201fd8 ; load "  " (inlined function starts here)
40201fff:	012d      		mov.n	a2, a1
40202001:	014b45        	call0	402034b8 <String::String(__FlashStringHelper const*)> ; call constructor for appended string
40202004:	013d      		mov.n	a3, a1
40202006:	0c2d      		mov.n	a2, a12
40202008:	015105        	call0	4020351c <String::concat(String const&)>  ; concatenate
4020200b:	012d      		mov.n	a2, a1
4020200d:	010605        	call0	40203070 <String::~String()> ; delete F("  ")
40202010:	21cb      		addi.n	a2, a1, 12
40202012:	0105c5        	call0	40203070 <String::~String()> ; delete arg1
40202015:	18c122        	addi	a2, a1, 24
40202018:	010545        	call0	40203070 <String::~String()> ; delete arg2
4020201b:	f108      		l32i.n	a0, a1, 60  ; adjust stack and return
4020201d:	0c2d      		mov.n	a2, a12
4020201f:	e1c8      		l32i.n	a12, a1, 56
40202021:	40c112        	addi	a1, a1, 64
40202024:	f00d      		ret.n
40202026:	200000        	or	a0, a0, a0
40202029:	fee8      		l32i.n	a14, a14, 60
4020202b:	3f          	.byte 0x3f
4020202c:	01c200        	slli	a12, a2, 32
```

So as you see compiler has decided to inline this function call. Furthermore, unused arguments are still initialized (because it is not clear to the compiler that they don't introduce side effects). Plus, there is no string pooling, so adding another call with same arguments will make extra copies of string literals you use as arguments.

The bytes in second column are actual instructions, so you can calculate how much each part contributes to the total size.

robertgregor, 2015-10-07:
Hello, thanks OK, your answer is excelent. Thanks! I see now, what is going on. so in total it takes 78 bytes of code. So as I see, there is no way, how to do it in better way. i.e. use pointers in the function arguments?

igrr, 2015-10-12:
```c++
String buildSubmitWithValue(__FlashStringHelper const* s1, __FlashStringHelper const* s2) {
    String ret = " ";
    return ret;
}
```
Might save you a few constructor calls.

closed 2015-10-17 by igrr