esp8266/Arduino #862: Add option to boards to remove the spiffs filesystem and increase the code space

robertgregor, 2015-10-06:
Hello, would it be possible to add to the boards.txt also:

generic.menu.FlashSize.512K=512K (0K SPIFFS)

robertgregor, 2015-10-07:
I did this and it is working:
generic.menu.FlashSize.512K64=512K (64K SPIFFS)
generic.menu.FlashSize.512K64.build.flash_size=512K
generic.menu.FlashSize.512K64.build.flash_ld=eagle.flash.512k.ld
generic.menu.FlashSize.512K64.build.spiffs_start=0x6B000
generic.menu.FlashSize.512K64.build.spiffs_end=0x7B000
generic.menu.FlashSize.512K64.build.spiffs_blocksize=4096
generic.menu.FlashSize.512K64.upload.maximum_size=434160

generic.menu.FlashSize.512K0=512K (0K SPIFFS)
generic.menu.FlashSize.512K0.build.flash_size=512K
generic.menu.FlashSize.512K0.build.flash_ld=eagle.flash.512k.ld
generic.menu.FlashSize.512K0.build.spiffs_start=0x6B000
generic.menu.FlashSize.512K0.build.spiffs_end=0x7B000
generic.menu.FlashSize.512K0.build.spiffs_blocksize=4096
generic.menu.FlashSize.512K0.upload.maximum_size=499696

igrr, 2015-10-07:
It may look like it works, but both options use the same linker script (`eagle.flash.512k.ld`). This means that the linker will use the same memory map for your 512K/0K variant, so when your code exceeds 434160, you will get linker error. You would also need to create another linker script (eagle.flash.512k0.ld) and adjust the values there.

igrr, 2015-10-26:
Added in 81566a9

asetyde, 2015-10-26:
Also I want this ! Thx

ahuillet, 2015-11-04:
igrr, I believe your SPIFFS address choice is incorrect: it breaks OTAs to set _SPIFFs_start to 0, because the start address for the update becomes 0 - 0x40200000 = way too high.

The following makes OTAs work:

-PROVIDE ( _SPIFFS_start = 0 );
-PROVIDE ( _SPIFFS_end = 0 );
+PROVIDE ( _SPIFFS_start = 0x4027B000 );
+PROVIDE ( _SPIFFS_end = 0x4027B000 );

closed 2015-11-30 by igrr

sven337, 2015-11-30:
You could go further: the SPIFFS library is pretty big in terms of code space.
In order to support OTAs on a 512kB ESP03, I had to remove this lib to ensure that the code would be small enough.
I don't know what the best way of doing this conditionally is, but I think it would be a valuable feature.

igrr, 2015-11-30:
@sven337 that would be a nice feature. Perhaps can be done by adding `-DDISABLE_SPIFFS` to compiler flags for the "no spiffs" variant.

igrr, 2015-11-30:
Actually this doesn't seem to be necessary, because linker does a good job at removing unused code.

When I build an empty sketch for Generic (64k SPIFFS), binary size is 198636 bytes. 
Running `xtensa-lx106-elf-nm -C out.elf | grep spiffs` returns no symbols.

When I add some FS functions, like this:
```c++
#include <FS.h>
void setup() {SPIFFS.begin();}
void loop() {}
```
Sketch size goes up to 229496 bytes. Running `nm` again, i see that spiffs library functions are present in the output file.

sven337, 2015-11-30:
This doesn't match my findings, on the contrary I had spiffs functions show up in the binary even without using any spiffs functions in the program. I'll check and get back to you.

asetyde, 2015-11-30:
Ok perfect . Thanks

sven337, 2015-11-30:
An empty sketch like you used does indeed not pull any spiffs functions, but one that includes ESP8266WebServer.h does.
It increases the code size by about 16kB. 

Interestingly, removing cores/esp8266/spiffs_api.cpp doesn't make the build fail, it simply removes this extra code that is useless when no spiffs is in use.

igrr, 2015-11-30:
Thanks, i think i have fixed that in 9e930e5. Code size is down from 250k to 228k for HelloServer sample.