esp8266/Arduino #863: mkspiffs not found!

irtwood, 2015-10-07:
Hello,
I am getting an error:
SPIFFS Error: mkspiffs not found! 
when using the ESP8266 Sketch Data Upload tool. 
I followed the instructions here: 
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/reference.md#uploading-files-to-file-system.

To resolve this issue I downloaded the mkspiffs.exe file from here https://ci.appveyor.com/project/igrr/mkspiffs/build/0.0.24/artifacts

However, I am unsure where to place the file to get this to work. I have looked at previous issues,
https://github.com/esp8266/Arduino/pull/775
and
https://github.com/esp8266/Arduino/issues/809

However, I do not see the mentioned \build\windows\work\hardware\ directory in arduino 1.6.5 on windows 10 (or in the sketch folder).
Could someone direct me to where I can find this, or where I should place this file?

Thank you very much!

igrr, 2015-10-07:
Which version of the core do you have installed? You can check it in Boards Manager:
[image]
It should say `1.6.5-1160-gef26c5f` in order for mkspiffs to be available.

irtwood, 2015-10-08:
Thank you! That was it. I had an older version installed. I changed the boards manager URL from:
http://arduino.esp8266.com/stable/package_esp8266com_index.json
to:
http://arduino.esp8266.com/package_esp8266com_index.json
Then reinstalled the package (now the same as the one shown). It works fine now. 

Thank you!

closed 2015-10-08 by igrr

MosabWadea, 2015-10-11:
hello @igrr 
I am facing this problem and I tried many solutions and it wasn't fixed..
I always have this version (1.6.5-947-g39819f0)
I tried the link provided from @irtwood  but it also didn't work
please help

spagr, 2015-10-17:
Try a link from staging version :  http://arduino.esp8266.com/staging/package_esp8266com_index.json

freeck, 2017-02-11:
I also had this bsp-version installed and got the same error message.
Updating the bsp-version via the boards manager did the job for me....