esp8266/Arduino #868: Remove the debug on serial port

robertgregor, 2015-10-08:
Hello, I have taken the latest build. Seems, there is small issue, thus I am not sure in which case, I will get on serial port this:

Received:
0, CB, 1, 9F, 6B =? 16B

From where it can be???

Links2004, 2015-10-09:
when this messages comes?
if its after a reset its normal, its the internal bootloader of the So there is nothing what we can change.
may using ```Serial.swap();``` is helpful for you application. (D13 + D15)

https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/reference.md#serial

[image]

igrr, 2015-10-17:
@robertgregor could you please post the sketch which demonstrates this issue? I don't think I'm seeing such output in latest builds.

robertgregor, 2015-10-21:
Hello, sorry for that, that debug was in DHT library not in your code.

closed 2015-10-21 by robertgregor