esp8266/Arduino #869: Random Exception (9) from Web Server

psYbR, 2015-10-09:
I frequently get this exception and subsequent reset of the ESP8266: http://i.imgur.com/gCp89u2.png

My code (https://github.com/psYbR/dual-esp8266-controller/blob/master/esp8266_controller.ino) is essentially just receiving HTTP requests and outputting the value to serial. The client (same .ino file, flashed to another ESP8266, just change SERVER to false) is another ESP8266 spamming HTTP requests at the server. The client is stable.

It can run for 5 minutes or only a few seconds before throwing the exception, it isn't consistent.

I am not running the latest staging however @luc-github is and can replicate the issue as per his comment on issue #428. I'll update to the latest staging and report any changes here

It may be worth mentioning on a couple of occasions I've seen different exceptions but as I'm not running the latest staging they may already be fixed (and I haven't been able to get them to occur again)

Cheers

psYbR, 2015-10-11:
Hi @igrr have you had a chance to have a look at this one? I am still getting this quite often

igrr, 2015-10-12:
I did take a look, but I wasn't able to reproduce as I don't have two ESPs with me currently. 
I've used your SERVER part and sent requests using netcat. Wasn't able to get it to crash, although it does run out of memory quickly. The reason for this is that your server side is doing active close of TCP connection, so the connection remains in TIME_WAIT state for two minutes per TCP spec.

psYbR, 2015-10-12:
ah I see. thanks @igrr! I'll alter the code not to close the connection each time and see if that fixes the crashing.

igrr, 2015-10-12:
In the case where you have ESP8266 on both sides (client and server), there isn't much you can do — one side will still have to do active close.
If you have ESP8266 on one side (server only), then you need to make sure client does active close. When using HTTP, there is "Connection: close" header that you may send.

Yet another option is to abort the connection by sending RST segment. There is `client.abort()` method for that. This avoids entering TIME_WAIT state on both sides, but is somewhat a hacky approach.

psYbR, 2015-10-13:
Thanks for the heads up :) Diverting a little away from ESP8266 questions and more towards HTTP in general I guess but is it possible to perform client.connect() once and then do client.print() each loop? Or do I need to perform a client.close() after each print? I am wanting to control a robot in realtime so the stream of data needs to be continuous and as fast as possible. Is this the best way to do it?

cheers

closed 2015-11-30 by igrr

andydarwin, 2015-12-25:
I got Similar Problem... I am making a program that can control LED through WIFI by combining Relay Control example from aRest.h 
and Live IP Camera with it.. so i can see the LED turned on or off directly

I frequently got This ERROR... Can anybody tell me Whats going on HERE ????

WiFi connected
Server started
192.168.1.222

Exception (9):
epc1=0x402153c6 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40214cc9 depc=0x00000000

ctx: sys 
sp: 3ffffde0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff80:  40214cad 3fffdab0 00000002 3ffefdc0  
3fffff90:  402146bc 3fffdab0 00000000 3fffdcc0  
3fffffa0:  3ffe9900 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
..
WiFi connected
Server started
192.168.1.222

eumb, 2016-01-30:
Same issue here.

Does anyone got a solution?

Links2004, 2016-01-30:
with out a objdump or at least a call stack analyses its hart to say where the error is coming from.
 - install the ESP Exception Decoder tool https://github.com/me-no-dev/EspExceptionDecoder
 - flash the ESP new from the IDE.
 - wait for stack dump
 - c&p the error to analyses 
 - post the result here

[image]

Pablo2048, 2016-02-20:
so here is one example. Hope that help...
[image]

Links2004, 2016-02-20:
can you try the same with latest git?
we have changed the heap management to umm.
since the SDK one make too much problems.

Pablo2048, 2016-02-20:
I can try, but I'm very beginner with Linux and I'm very happy that Arduino with ESP8266 run, so it will take some time ;-)

Pablo2048, 2016-02-20:
Ok, so I installed actual git version. I'm doing heavy testing right now (last error took me about 1/2 hour to generate...).

Pablo2048, 2016-02-20:
so here is the result:
Uptime 0 days 00:56:06.628
Requests 1600 (2 seconds autorefresh)
It seems like this error is gone. Thank You Links2004.

jeferrb, 2016-03-03:
If I'm sure, I'm getting the same problem as Pablo.
I'm using the last git version.
Have someone any advice?

There are three samples below (split by count.S:109):
[image]

igrr, 2016-03-03:
Hi @jeferrb, what makes you think the exception is related to WebServer? I don't see a single stack frame from ESP8266WebServer library in the screenshot you've posted.

jeferrb, 2016-03-03:
Hi @igrr, thanks for reply. It is really not a WebServer question. Since it stops on the same line I thought it's related. I'm trying to port a project that works on Arduino Mega and I need some hint to understand where these error come from.

igrr, 2016-03-03:
The final line in stack trace is actually the first function executed. Stack traces are normally read from bottom to top. I.e. the last function is actually `GetRandomBitRaw2`.
You might want to read https://github.com/esp8266/Arduino/blob/master/doc/exception_causes.md to get better understanding of exception line at the top of stack trace.

jeferrb, 2016-03-03:
Hum, I get it now. So many thanks igrr, you help a lot. And, you are doing a grate job.

bamboklaat, 2016-06-19:
Hi everyone, sort of late, but I found myself in a problem related to this one (I think). I started with arduino UNO since I found the loading much faster that ESP, so it was ok untill I started to migrate, my sistem is a LCD & rtc1307 (I2c) and a rotary encoder (with push bottom). My 1st step was to make the 3 periphericals work ok on the ESP, and it did work, just a few modification needed, the point was to make the whole code work ok, I got the EXEPTION (9) (Load or store to an unaligned address) when the interrupt take place. First thing I did was to remove the "lcd.createChar(0, grado); "  believing that the "address 0" was causing problems, but the problem remains, also tried a small script to check it out, and this character appeared on the screen, so it doesnt seems to be that. later on, tried the "TIME" pin in the 3 possible states, when low and high, the LCD stays frozen in the main manue, leaving the time pin floating, at least it interrupts, goes to the main manu (showing the current time from rtc) and then fades away as it should.

Unfortunally, with my knowledge I can not go further than what I tried, any suggestion? I paste the code in case you find a critic variable around. Thanks a lot for reading.

`define encoder0PinA 13
#define encoder0PinB 12
#define buttonPin 14

// Librerias necesarias
#include "Wire.h"
#include "LiquidCrystal_I2C.h"
#include "DS1307new.h"
#include <Wire.h>  // Comes with Arduino IDE
#include <LiquidCrystal_I2C.h>

LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Set the LCD I2C address
byte grado[8] = {B11100, B10100, B11100, B00000, B00000, B00000, B00000, B00000};           //Simbolo °
byte termometro[8] = {B00100, B01010, B01010, B01110, B01110, B11111, B11111, B01110};      //icon for termometer
byte hume[8] = {B00100, B00100, B01010, B01010, B10001, B10001, B10001, B01110};
byte bar[8] = {B11111, B11111, B11111, B11111, B11111, B11111, B11111, B11111};

int t, h;

#define MON_BUF_LEN 128

String mes, dia, anno;
String hora, minuto, segundo;
String vacia, horaChar, fechaChar;
char horaC[10], fechaC[11];
int second_old, minute_old, hour_old, monthDay_old, month_old, year_old;

char mon_buf[MON_BUF_LEN];
uint8_t mon_buf_cnt = 0;
char *mon_curr;

uint8_t mon_is_date;
uint16_t mon_year;
uint8_t mon_month;
uint8_t mon_day;

uint8_t mon_is_time;
uint8_t mon_hour;
uint8_t mon_minute;
uint8_t mon_second;
uint8_t mon_error;

unsigned long currentMillis;
unsigned long inter_backlight = 8000;      // va hasta 32K = 32 seg
unsigned long m;

volatile int lastEncoded = 0;
volatile long encoderValue = 0;
bool CW = 0;
int encoded;

int vel_vent = 50;
int var_hora_ini = 6;
int var_Periodo = 18;

long lastencoderValue = 0;

int lastMSB = 0;
int lastLSB = 0;

int encoder = 11;     // 10 u 11, defino el arranque

int MenuBoxTop = 2;          // Number of items in the Send Menu update this  if you change the number of menu items
int cityBoxTop = 4;       // if you change the submenu you have the changes this value as well
int HoraBoxTop = 3;

[... 663 lines cut ...]
  {
    RTC.getRAM(addr, ram, 8);
    mon_print_3d(addr);
    Serial.print(": ");
    for ( i = 0; i < 8; i++ )
    {
      mon_print_3d(ram[i]);
    }
  }
}

void mem_memory(void)
{
  uint8_t a, m;
  a = mon_get_value();
  m = mon_get_value();

}

void exec(void)
{
  //  Serial.println(mon_buf);
  mon_curr = mon_buf;

}`

MHz000, 2016-10-10:
as a newbie I've the same problem and **no idea how to proceed  :-((**  

Exception (9):
epc1=0x40202846 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe8542 depc=0x00000000

ctx: cont
sp: 3ffefa00 end: 3ffefc10 offset: 01a0

>>>stack>>>
3ffefba0:  3ffeeab0 3ffee4b0 0000003a 40202866 
3ffefbb0:  3fffdad0 3ffee4b0 3ffeebb8 40201da5 
3ffefbc0:  00000000 00000000 00000000 40202c78 
3ffefbd0:  00000000 feefeffe 3ffeebb8 40202c9c 
3ffefbe0:  00000000 00000000 00000001 3ffeebe4 
3ffefbf0:  3fffdad0 00000000 3ffeebdc 40203150 
3ffefc00:  feefeffe feefeffe 3ffeebf0 40100718 
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

after one or two successfull requests.

It woud be fine if someone could make a little test on his system in his envireoment to see it something is wrong in my setup.
Here is the simplified code
//--------------------------------------------------------------------------------------------------------------------------------------------------------------------
char versions_id[] = "ESP__WEB_test_V6";
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ArduinoJson.h>
#include <TimeLib.h>

const char WIFI_SSID[] = "...";               // WiFi information
const char WIFI_PSK[] = "...";

const char http_site[] = "api.usno.navy.mil";       // Remote site information
const int http_port = 80;

const int LED_PIN = 16;                                   // global variables & definitions
WiFiClient client;
char c;                                                              // incomming char from WEB-server
unsigned long delayMillis;                               // delay loop time
unsigned long pause = 20000;                        // prog paused

// JSON stuff
StaticJsonBuffer<1000> jsonBuffer;                  // Memory pool for JSON object tree.
char json[500];                                     // JSON object char array, filld with server response
String head = "";                                   // HTTP header from server
int json_i=0;                                       // index for json char array
int JS_bracket = 0;                                 // counter for JSON { and } brackets 

void setup() {
  Serial.begin(115200);                             // Set up serial console to read web page
  Serial.println(); Serial.println(versions_id);
  Serial.println("********************RE-/START******************");
  pinMode(LED_PIN, OUTPUT);                         // Set up LED for debugging
}

void loop() {
  head = "";                                        // reset server response string
  JS_bracket = 0;                                   // reset json bracket counter
 
  connectWiFi();                                    // Connect to WiFi
 
  if ( !getPage() ) {                               // Attempt to connect to website   
    Serial.println("GET request failed");}
    else {
    Serial.println("GET request ok");}
   
    while (client.connected() ) {
    if ( client.available() ) {                     // If there are incoming bytes, print them
      c = client.read();
      //Serial.print(c);
      if (JS_bracket == 0) {head=head+c;}           // fill in server response
      if (c == '{') {++JS_bracket;}                 // ectraxt json object
      if (c == '}') {--JS_bracket; }
      // still to do: skip white space outside ".."
      if (JS_bracket > 0) { //Serial.print(c);      // look for matching closing bracket
      json[json_i]=c; json_i++;}; }
    }
    json[json_i]='}';                               // add a closing } bracket 
    Serial.println("stop client");                  // If the server has disconnected, stop the client and WiFi
    client.stop();                                  // Close socket and wait for disconnect from WiFi
    if ( WiFi.status() != WL_DISCONNECTED ) {
      WiFi.disconnect();}
    digitalWrite(LED_PIN, HIGH);                    // Turn off LED
    
// *****process server data here
  Serial.println("processing server data");
//  for (int i=0; i<=json_i; i++) {Serial.print(json[i]);}  Serial.println();  // control print
  Serial.println("******************pausing****************");
  delayMillis= millis()+pause;                             // 5 minutes intervall 
   do {
     yield();
     } while (millis() < delayMillis);    
  Serial.println("***********start after pausing***********");
// *****end processing server data  
  
}

//******************connect WiFi***********************
void connectWiFi() {                            // connect WiFi
byte led_status = 0;
WiFi.mode(WIFI_STA);                            // Set WiFi mode to station (client)
WiFi.begin(WIFI_SSID, WIFI_PSK);                // Initiate connection with SSID and PSK
while ( WiFi.status() != WL_CONNECTED ) {       // Blink LED while we wait for WiFi connection
  yield();
  digitalWrite(LED_PIN, led_status);
  led_status ^= 0x01;
  delay(100);
  }
digitalWrite(LED_PIN, HIGH);                    // Turn LED on when we are connected
}

//***********************get page***************
bool getPage() {                                // Perform an HTTP GET request to a remote page
if ( !client.connect(http_site, http_port) ) {  // Attempt to make a connection to the remote server
return false;
}
client.println("GET /rstt/oneday?date=9/14/2016&coords=47.73N,9.16E&tz=1 HTTP/1.1"); // Make an HTTP GET request
client.print("Host: ");
client.println(http_site);
client.println("Connection: close");
client.println();
return true;
}
//------------------------------------------------------------------------------------------------------------------------
enviroment:  WIN10 arduion 1.6.12

Thank you for any help or hint

borqborq, 2018-08-20:
I had the same problem: Exception (9). My issue was that the page would render for the client, but you couldn't submit the tiny form that was on that page. It threw the error every time. 

A 'tremor in the force' felt like it was memory related - I thought maybe the array capturing the clients HTTP response was too big and popping the RAM so I kept trimming the array variable allocation down with no love...

On a whim, I made the array size bigger and the error vanished. The 8266 web server runs smooth 99.9% (because nothing is ever 100% in IOT, right?)

It turned out that the array that was capturing the HTTP_RESPONSE with was not big enough (by a hair). I originally allocated an array size of 500, which was enough to allow the web client to load or refresh the page successfully, but if there was any other data added from the client and sent to the 8266 (i.e. a tiny web form on the page), it would throw the error after submitting the form.

`char http_resp[500] = {0};`

After changing the array size to 700, the error was gone, and the 8266 takes the form submission and processes it just fine.

`char http_resp[700] = {0};`

Long story short, my problem was in the handling of the arrays and HTTP responses from the client.

Hope this helps someone out there!

rajsingha, 2019-12-14:
> ah I see. thanks @igrr! I'll alter the code not to close the connection each time and see if that fixes the crashing.

Please tell me how did you fixed it?

johny77g, 2020-10-25:
Hi, had similar problem with exception 9. Finally in BoardsManager in esp8266 I installed older version 2.3.0 - additionally I had to choose Board Wemos D1 retired - and problem with exception 9 disappeared completely.

riskiadi, 2020-10-26:
Exception (9):
20:33:41.692 -> epc1=0x4020b648 epc2=0x00000000 epc3=0x40000f68 excvaddr=0x00000041 depc=0x00000000

after add firebase database library and arduinojson version 5 latest

devyte, 2020-10-26:
I understand that firebase uses WiFiClientSecure, research that and you'll know what's wrong.

bappapan220, 2021-05-15:
Thank you Sir. thanks a lot.