esp8266/Arduino #870: ESP8266 default power mode

shirish47, 2015-10-09:
what is the default power mode based on this chart.
http://bbs.espressif.com/download/file.php?id=177&sid=ece47e1a03b85c5a4c9f395381d56e7a

Please and how can we change it.

asetyde, 2015-11-04:
Depends what are you doing with esp ... If you use wifi with modern router you can see 2 value , tx and rx with N type . but this are oscillating value between 50/120 milliampere i think .
you connect a sensor 100ma ? 220ma max and 150ma min .. or you can use a tester

shirish47, 2015-11-04:
thanks ..I mean't those set in this library ... But we later found the setting git..

asetyde, 2015-11-04:
ook

igrr, 2015-11-04:
@shirish47 so is this question resolved for you? If so, could you please mark it closed?

closed 2015-11-09 by shirish47