esp8266/Arduino #871: [enhancement] key value storage system/library

tzapu, 2015-10-09:
a way to easily save key value pairs of variables
ideally this would also have some wear levelling features included

mentioned here
https://github.com/esp8266/Arduino/issues/849#issuecomment-146674974

pgollor, 2015-10-11:
maybe you are looking for this:
https://github.com/maniacbug/StandardCplusplus/

There is a map part like the std CPP library.

drmpf, 2015-10-11:
http://thijs.elenbaas.net/2012/07/extended-eeprom-library-for-arduino/
provides a general eeprom storage for various types

tzapu, 2015-10-12:
hi guys, great suggestions
i have added the issues as a reminder for igrr's suggestion of having such a thing maybe in the core. something that i assume won t need addresses and so on, just save(key, value) retrieve(key), and that would of course be piped through to eeprom or fs, without the actual need of those libraries or others to be included.
just something to get you started asap with some persistent storage

devyte, 2017-10-20:
I think [ArduinoJson](https://github.com/bblanchon/ArduinoJson) meets this requirement, if used with a single level of nesting.
As for wear leveling, SPIFFS does wear leveling under the hood, so storage as a file on SPIFFS accomplishes that.
Closing.

closed 2017-10-20 by devyte