esp8266/Arduino #874: SPI LIbrary Does Not Honor Clock Polarity Setting

YorkBW, 2015-10-09:
Essentially, SPI modes 0 and 1 are unsupported; if these modes are requested, you actually get modes 2 and 3 respectively. It's noted in the source that setting the hardware SPI clock polarity is "TBD," but I don't think that's documented anywhere outside the source.

igrr, 2015-10-09:
This is mentioned in reference:

https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/reference.md#spi

YorkBW, 2015-10-09:
Thanks! Although that document says modes 2 and 3 aren't supported, while I think it's the other way round. I had to set an external commercial module to accept mode 2 for it to work with the ESP8266.

YorkBW, 2015-10-09:
After my too-brief research, I'm suspecting that playing around with values for SPI_CLKCNT_H and SPI_CLKCNT_L might be a way to flip the clock polarity; unfortunately I don't have the toolchain installed right now to test and contribute. And hey -- thanks for making this code available, it has saved me a ton of time!!!

igrr, 2015-10-12:
If you got ESP8266 Arduino core installed, then you do have a toolchain installed. You can experiment with those settings by adding necessary .h/.cpp files to the sketch.

Duality4Y, 2015-10-19:
I have done research on implementing this, and I am not sure if the esp8266 hardware supports it. at-least it doesn't show in the research I did.

igrr, 2016-03-23:
Fixed in #1781

0xPIT, 2016-03-26:
I've tested just now with my logic analyzer. Unfortunately, it does **not** work.
I've also tried all modes (0,1,2,3) and SCK always idles in low state, whereas with CPOL=1 (in modes 2 & 3) it should idle at high.

 @Duality4Y where did you find the info about bit 29 in SPI1P is setting CPOL?

0xPIT, 2016-03-26:
Also, contrary to the initial report from @YorkBW, actually only modes 0 and 1 are supported, as the clock always idles at low. Modes 2 & 3 would require the clock idling in high state, which does not work.

See here: https://en.wikipedia.org/wiki/Serial_Peripheral_Interface_Bus#Clock_polarity_and_phase

0xPIT, 2016-03-26:
OK, I'm wrong. The CPOL Setting works, but my problem might be something different: Obviously, when I initialize with SPI_MODE0, the mode can not be changed later.

Duality4Y, 2016-03-28:
@0xPIT I think you didn't meen to mention me?

igrr, 2016-03-28:
@0xPIT Here are the only references I found:
https://github.com/UncleRus/esp-open-rtos/blob/core/spi/core/esp_spi.c#L73-L91
http://bbs.espressif.com/viewtopic.php?t=342#p5384

Also linking #1751 here.

igrr, 2016-03-28:
BTW, as it is [pointed out](https://github.com/UncleRus/esp-open-rtos/blob/core/spi/core/esp_spi.c#L77-L78), in the source linked above, CPHA needs to be inverted when CPOL == 1. I think this needs to be fixed in our code as well.

closed 2016-04-18 by igrr