esp8266/Arduino #876: ESP8266 in Arduino IDE - CRC error with board manager

PiRiFr, 2015-10-10:
Hello,

I have a working Arduino IDE1.6.5 (French language) under windows7.
I am trying to load the ESP8266 board with the board manager.
I have placed the URL:  http://arduino.esp8266.com/stable/package_esp8266com_index.json
In the additional boards manager URL.
Then when selecting INSTALL for the ESP8266 board, a download happens and eventually ends up with the error: 
                                    "CRC doesn't match. File corrupted"

Help is welcomed, all I tried failed!
PS I tried both stable and beta versions with the same result.

Juppit, 2015-10-10:
Rename C:\Users\[username]\AppData\Roaming\Arduino15 to something like 
Arduino15.sav and try again. When ok, you may delete the directory 
Arduino.sav.

On 10.10.2015 at 14:52 wrote PiRiFr:

PiRiFr, 2015-10-17:
Hi Juppit,

Simple and easy !
Thank you very much, everything is working nicely now.

closed 2015-10-17 by igrr

BelalMohsen, 2016-07-11:
Thank you Juppit

raghavenderlomte, 2016-08-30:
even after changing arduino to arduino.sav,i am facing the same problem

AgustinPelaez, 2016-10-19:
I had the same problem during a training. We were behind a company proxy that was somehow blocking  the libraries download. Switched to a regular network (no proxy no firewall) and worked perfectly.

DrSaad77, 2021-02-19:
Hi, I have the same problem! Trying to download ESP8266 in Board Manager, and it always ends with this CRC error.
I tryed to download it more than 10 times already, and no succes.
What I tried:
Renaming Arduino15 >> Arduino.sav, and when I re-open IDE, it again recreate its folder Arduino15.
Tried again, as it seems pure default setting, still error CRC
I even disabled AV soft, still error (((
Are there any news solution please.
Thx

batdog1981, 2022-03-08:
ive also tried re-installing tried re-naming, tried different versions of IDE, adding fully downloaded package to packege folder, tried different IDE but it wouldnt let me add any kind of esp8266 board. please help

DrSaad77, 2022-03-08:
> I've also tried re-installing tried re-naming, tried different versions of IDE, adding fully downloaded package to package folder, tried different IDE but it wouldn't let me add any kind of esp8266 board. please help. I'm not new to Arduino I have good grasp on the IDE but not enough to trouble shoot on my own

Only solution I can give you, is to uninstall totally IDE, then clean everything related to IDE and Arduino, even folder located in MyDocuments, and also in AppData\Local\Arduino15. Delete everything, use CCleaner (some will say it is placebo...). Then restart, and reinstall everything.
Install IDE, and update the Library, and also the board, then add the URL:
http://arduino.esp8266.com/stable/package_esp8266com_index.json

And let the package get installed smoothly, do not rush anything. 
Arduino IDE is very poor on the topic of managing its package, and everything.
Hope, it ll work, good luck ))

RuizMSebas, 2022-03-09:
> I've also tried re-installing tried re-naming, tried different versions of IDE, adding fully downloaded package to package folder, tried different IDE but it wouldn't let me add any kind of esp8266 [...]

I agree with DrSaad77, maybe a fresh install of the IDE will help. I've found [this forum](https://forum.arduino.cc/t/error-code-crc-doesnt-match/589224) that could give you a hint of some other things that you could try.

I've had similar issues with the IDE when trying to install packages and libraries. Usually a complete re-install of everything fixes the issue.

DrSaad77, 2022-03-09:
> I agree with DrSaad77, maybe a fresh install of the IDE will help. I've found [this forum](https://forum.arduino.cc/t/error-code-crc-doesnt-match/589224) that could give you a hint of some other [...]

I would like to add something.
From yesterday I am trying to find out which package of esp8266 by community, is installed, and everytime I type "esp8266", which is supposed to show me only esp8266 related, and by magic nothing!!! Not even esp8266 by community, which I am sure is installed on my PC. Maybe some error related to server or dont know what...
As I said, IDE is very poor in library managing, always error showing exactly what someone is looking for, or even sometime, you select "Possible to get updated", and it show you all packages!!
This kind of trouble always happens to me, and after some headaches, I clean everythig, and reinstall from scratch...
Most disturbing, is that you need to re-install again all library that you gona need...