esp8266/Arduino #878: How to access raw POST request content

Diaoul, 2015-10-11:
Sorry if this is a dumb question I couldn't find any mention of this in the docs or examples.
Using ESP8266WebServer, I want to process the raw content of an incoming POST request.

Is that possible?

Duality4Y, 2015-10-11:
How do you meen the raw content of the POST request ?

Diaoul, 2015-10-11:
Well, the data POST, after the headers. I'm not using `application/x-www-form-urlencoded` but json.

Diaoul, 2015-10-11:
After digging in the source code of the library I found the answer, raw content is considered as an argument named "plain".

I leave the issue open because this is quite unusual and should be documented somewhere.

me-no-dev, 2015-10-22:
what do you think is the proper way to do it?

Diaoul, 2015-10-22:
RFC defines the content of a request as the "message-body" (see https://tools.ietf.org/html/rfc7230#section-3)
I think "content" is also a good name and is used in the popular python library [requests](http://docs.python-requests.org/en/latest/user/quickstart/#response-content)

me-no-dev, 2015-10-22:
I vote for "messageBody" :) or just "body"

marvinroger, 2015-11-06:
I also vote for `body`, the message part is implicit. It is also a name used by popular libraries, like the `request` module in Node.js

dave-prosee, 2016-05-15:
Another vote for getting this documented. Thanks for posting the issue and your findings; it helped me.

tinxx, 2022-01-29:
This issue can be closed. The documentation is available in the [README](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/README.rst).

> ``arg`` - get request argument value, use ``arg("plain")`` to get POST body

closed 2022-01-29 by d-a-v