esp8266/Arduino #881: I want to use microseconds in timer interrupt like in timerone.h for ESP8266

GerdHeber, 2015-10-12:
Hello,
I've found ticker.h, but this library uses milliseconds, I want to work a little faster. I tried to set the PWM frequency, this works, but I want to count something at every pulse, so an interrupt would be best.
greetings
Gerd

igrr, 2015-10-12:
timer1 functions in Arduino.h:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h#L88-L109

See https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_pwm.c for usage example.

GerdHeber, 2015-10-12:
Hello Igor,
thanks a lot, after some try's it's working. One question I have... What ist TIM_EDGE and TIM_LEVEL doing? With TIM_EDGE my sketch is working, with TIM_LEVEL not.
greetings
Gerd

igrr, 2015-10-13:
Timer1 (FRC1 in Espressif's terms) has an edge-triggered interrupt, so I
guess there is no way to make TIM_LEVEL to work. Timer0 (based on
CCOUNT/CCOMPARE registers) is level triggered.

closed 2015-10-17 by igrr

sohadfield, 2015-12-06:
Hello Igor,

The links that you posted above for the timer1 functions are broken.  Can you update them please?

Thanks,

steve