esp8266/Arduino #882: Serial.println - print more then need?

gorec2005, 2015-10-12:
Hi!
i use https://github.com/milesburton/Arduino-Temperature-Control-Library
May be i misunderstand, but this code:
```
#include <OneWire.h>
#include <DallasTemperature.h>

#define ONE_WIRE_BUS 2
OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);
float temp_prev[8] = {0,0,0,0,0,0,0,0};
char* addrs[8] = {"","","","","","","",""};

void setup(void)
{
  Serial.begin(9600);
  sensors.begin();
}

void loop(void)
{ 
  int dc = sensors.getDeviceCount();
  sensors.requestTemperatures();
  for (int i=0; i<dc; i++) {
     float temp = sensors.getTempCByIndex(i);
     if (temp != temp_prev[i]){
       DeviceAddress da;
       sensors.getAddress(da,i);
       sprintf(addrs[i],"[%X%X%X%X%X%X%X%X]",da[0],da[1],da[2],da[3],da[4],da[5],da[6],da[7]);
       Serial.println("{"+String(addrs[i])+"} index "+String(i)+" is: "+String(temp)+"!");
       temp_prev[i]=temp;
     }
  }
  delay(5000);
}
```

print ro serial this:

```
{[28FFD5163140BB]} index 0 is: 24.50!
[28FFD5163140BB]{[28FFC31C314040]} index 1 is: 24.44!
[28FFC31C314040]{[28FFC31C314040]} index 1 is: 24.50!
[28FFC31C314040]{[28FFC31C314040]} index 1 is: 24.44!
[28FFC31C314040]{[28FFD5163140BB]} index 0 is: 24.44!
```

it's a bug in serial.println or my fault in code? please help...

igrr, 2015-10-12:
`char* addrs[8] = {"","","","","","","",""};`
`sprintf(addrs[i],..`
You should reserve buffers for sprintf with correct size, instead of sprintf-ing into a one-byte buffer.

gorec2005, 2015-10-12:
oh - it's my fault... sorry

closed 2015-10-12 by gorec2005