esp8266/Arduino #884: flash size other than 512kb for standart esp8266 (esp-01) module

gorec2005, 2015-10-12:
Please explain - how i can make new partitions for 8MB for standart esp8266 module

Links2004, 2015-10-12:
the SDK not allow access over 4MB, if you add a 8MB chip you can only use the lower 4MB.
all needed files for 8MB and 16MB are there but disabled. thanks to the limitations from espressif.
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/boards.txt#L132

gorec2005, 2015-10-12:
:-(

gorec2005, 2015-10-12:
But may be use boot loader from http://richard.burtons.org/2015/05/18/rboot-a-new-boot-loader-for-esp8266/ ?
it's "can use the full size of the SPI flash" ... "8MBit of flash can be memory mapped at a time (the SDK bootloader allows at most the first 2 x 8Mbit chunks to be used for roms, rBoot doesn’t have this limit, on a 32MBit flash you can have 4 x 8MBit roms)"

Links2004, 2015-10-12:
32MBit --> 4MB
same limitation that we already have? or do i miss some point.

closed 2015-10-13 by gorec2005

douglasresende, 2015-10-31:
https://github.com/esp8266/Arduino/blob/master/boards.txt#L132