esp8266/Arduino #885: Arduino uses the stock Wire.h library instead of the ESP8266 Version

karpour, 2015-10-12:
Arduino 1.6.5, latest staging version.
Arduino prefers using the stock Wire.h even when Board is set to ESP8266. Not sure if this is an Arduino issue or if there is anything additional to do to convince Arduino to use the ESP8266 library.

roccomuso, 2015-10-13:
Do you think this could be related to the issue i'm currently having: https://github.com/esp8266/Arduino/issues/887 ?

pgollor, 2015-10-14:
Do you have a other library namend Wire in your library dir in your sketch workspace?

This could be the Problem. If i remember correctly, arduino prefers the library in your sketchfolder.

igrr, 2015-10-17:
@karpour please enable verbose compiler output in Arduino preferences and paste the whole output. First few lines should tell us what's wrong with your setup.

closed 2015-11-15 by igrr