esp8266/Arduino #887: NodeMCU v1.0 devkit - I2C issue (arduino 1.6.5)

roccomuso, 2015-10-13:
Hi guys,
I'm using a Nodemcu v1.0 (ESP-I2E) to communicate with a MPU-6050 accelerometer through the I2C.
But communication is not working. I Tried it on the Arduino Uno, so the accelerometer MPU-6050 works.

This is the code I'm trying to execute:

    #include<Wire.h>

    const int MPU=0x68;  // I2C address of the MPU-6050
    int16_t AcX,AcY,AcZ,Tmp,GyX,GyY,GyZ;
    void setup(){
      Wire.begin(1, 2); // sda, scl
      Wire.beginTransmission(MPU);
      Wire.write(0x6B);  // PWR_MGMT_1 register
      Wire.write(0);     // set to zero (wakes up the MPU-6050)
      Wire.endTransmission(true);
      Serial.begin(115200);
      Serial.println("Setup completato...");
    }
    
    void loop(){
      Wire.beginTransmission(MPU);
      Wire.write(0x3B);  // starting with register 0x3B (ACCEL_XOUT_H)
      Wire.endTransmission(false);
      Wire.requestFrom(MPU,14,true);  // request a total of 14 registers
      AcX=Wire.read()<<8|Wire.read();  // 0x3B (ACCEL_XOUT_H) & 0x3C (ACCEL_XOUT_L)    
      AcY=Wire.read()<<8|Wire.read();  // 0x3D (ACCEL_YOUT_H) & 0x3E (ACCEL_YOUT_L)
      AcZ=Wire.read()<<8|Wire.read();  // 0x3F (ACCEL_ZOUT_H) & 0x40 (ACCEL_ZOUT_L)
      Tmp=Wire.read()<<8|Wire.read();  // 0x41 (TEMP_OUT_H) & 0x42 (TEMP_OUT_L)
      Serial.print("AcX = "); Serial.print(AcX);
      Serial.print(" | AcY = "); Serial.print(AcY);
      Serial.print(" | AcZ = "); Serial.print(AcZ);
      Serial.print(" | Tmp = "); Serial.println(Tmp/340.00+36.53);  //equation for temperature in degrees C from datasheet
     
      delay(333);
    }

I'm supposing the only thing that i need to change are the pins (SDA and SCL). Following the board pinmap ( https://raw.githubusercontent.com/nodemcu/nodemcu-devkit/master/Documents/NODEMCU-DEVKIT-INSTRUCTION-EN.png ) I made a lot of attempts:
SDA = 14, SCL = 12
SDA = 1, SCL = 2
andat last:
SDA = 5, SCL = 6
With the last one actually the script doesn't even works! I'm having this error printed on the console:

    ets Jan  8 2013,rst cause:4, boot mode:(3,6)
    
    wdt reset
    load 0x4010f000, len 1264, room 16 
    tail 0
    chksum 0x42
    csum 0x42
    ~ld

Can someone tell me what are the right pins or what am i doing wrong? :/

RattyDAVE, 2015-10-13:
I am not sure but it looks like to me 

In the line 

'''
Wire.begin(1, 2); // sda, scl
'''

Change to

'''
Wire.begin(D1, D2); // sda, scl
'''

Also Default ...

GPIO5	D1	SCL
GPIO4	D2	SDA

roccomuso, 2015-10-13:
wow dude, thank you! this way it works:

    Wire.begin(4, 5); // sda, scl

With sda cable connected to D2 and scl cable connected to D1.

Why on the devkit pinamp there's no reference on the D2 and D1 pins? They are used for I2C, am i right?

Links2004, 2015-10-14:
I2C is implemented in software you can map it to almost all pins you like.

roccomuso, 2015-10-14:
So even a ESP8266-01 can use I2C?

mozgy, 2015-10-14:
Yes, it can, check www.esp8266.com, there's plenty of examples ..

igrr, 2015-10-17:
@roccomuso is this issue resolved for you?

roccomuso, 2015-10-17:
Yes thank you @igrr

closed 2015-10-18 by igrr

ketancool203, 2016-01-06:
I am also trying to connect mpu6050 with esp8266. But facimg problem please help.
wanted to send MPU6050 data to a IP ADDRESS.

waqa123, 2016-07-16:
I'm having same problem. I've tried your solution but it's not working... I'm tryng to interface mpu6050(GY-510) with Esp8266 nodeMCU v3. My code is shown below. Which RAW example of JeffRob. But all the raw values are coming out Zero which means there is a problem of communication(I2C)  between GY-510 and NodeMCU V3. Plz help me out of this.

 #include <Wire.h>
// I2C device class (I2Cdev) demonstration Arduino sketch for MPU6050 class
// 10/7/2011 by Jeff Rowberg <jeff@rowberg.net>
// Updates should (hopefully) always be available at https://github.com/jrowberg/i2cdevlib
//
// Changelog:
//      2013-05-08 - added multiple output formats
//                 - added seamless Fastwire support
//      2011-10-07 - initial release

/* ============================================
I2Cdev device library code is placed under the MIT license
Copyright (c) 2011 Jeff Rowberg

Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to deal
in the Software without restriction, including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in
all copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
THE SOFTWARE.
===============================================
*/

// I2Cdev and MPU6050 must be installed as libraries, or else the .cpp/.h files
// for both classes must be in the include path of your project
#include "I2Cdev.h"
#include "MPU6050.h"

// Arduino Wire library is required if I2Cdev I2CDEV_ARDUINO_WIRE implementation
// is used in I2Cdev.h
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
    #include "Wire.h"
#endif

// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for InvenSense evaluation board)
// AD0 high = 0x69
MPU6050 accelgyro;
//MPU6050 accelgyro(0x69); // <-- use for AD0 high

int16_t ax, ay, az;
int16_t gx, gy, gz;

// uncomment "OUTPUT_READABLE_ACCELGYRO" if you want to see a tab-separated
// list of the accel X/Y/Z and then gyro X/Y/Z values in decimal. Easy to read,
// not so easy to parse, and slow(er) over UART.
#define OUTPUT_READABLE_ACCELGYRO

// uncomment "OUTPUT_BINARY_ACCELGYRO" to send all 6 axes of data as 16-bit
// binary, one right after the other. This is very fast (as fast as possible
// without compression or data loss), and easy to parse, but impossible to read
// for a human.
//#define OUTPUT_BINARY_ACCELGYRO

#define LED_PIN 13
bool blinkState = false;
[... 56 lines cut ...]

    #ifdef OUTPUT_READABLE_ACCELGYRO
        // display tab-separated accel/gyro x/y/z values
        Serial.print("a/g:\t");
        Serial.print(ax); Serial.print("\t");
        Serial.print(ay); Serial.print("\t");
        Serial.print(az); Serial.print("\t");
        Serial.print(gx); Serial.print("\t");
        Serial.print(gy); Serial.print("\t");
        Serial.println(gz);
    #endif

    #ifdef OUTPUT_BINARY_ACCELGYRO
        Serial.write((uint8_t)(ax >> 8)); Serial.write((uint8_t)(ax & 0xFF));
        Serial.write((uint8_t)(ay >> 8)); Serial.write((uint8_t)(ay & 0xFF));
        Serial.write((uint8_t)(az >> 8)); Serial.write((uint8_t)(az & 0xFF));
        Serial.write((uint8_t)(gx >> 8)); Serial.write((uint8_t)(gx & 0xFF));
        Serial.write((uint8_t)(gy >> 8)); Serial.write((uint8_t)(gy & 0xFF));
        Serial.write((uint8_t)(gz >> 8)); Serial.write((uint8_t)(gz & 0xFF));
    #endif

    // blink LED to indicate activity
    blinkState = !blinkState;
    digitalWrite(LED_PIN, blinkState);
}

MuseDNA, 2016-11-09:
I'm using a Nodemcu v1.0 (ESP-I2E) to communicate with i2c Pin There are external arduino?

ahmadSum1, 2018-06-10:
I am also facing issues using GY-521 (MPU6050) with esp12e. 
My issue is that it really unstable data in esp while works perfectly with arduino uno/nano. Not a library issue since I'm using direct wire library functions.
Here is the code:

```
  ///////////////////////////////////////////////////////////////////////////////////////
  //Source
  ///////////////////////////////////////////////////////////////////////////////////////
/*  Website: http://www.brokking.net/imu.html
  Youtube: https://youtu.be/4BoIE8YQwM8
  Version: 1.0 (May 5, 2016)

  ///////////////////////////////////////////////////////////////////////////////////////
  //Connections
  ///////////////////////////////////////////////////////////////////////////////////////
  Power (5V) is provided to the Arduino pro mini by the FTDI programmer

  Gyro - Arduino pro mini
  VCC  -  5V
  GND  -  GND
  SDA  -  A4
  SCL  -  A5

  LCD  - Arduino pro mini
  VCC  -  5V
  GND  -  GND
  SDA  -  A4
  SCL  -  A5
*//////////////////////////////////////////////////////////////////////////////////////

///////in NodeMCU test setup
#define scl D5
#define sda D6

//Include LCD and I2C library
#include <LiquidCrystal_I2C.h>
#include <Wire.h>

//Declaring some global variables
int gyro_x, gyro_y, gyro_z;
long acc_x, acc_y, acc_z, acc_total_vector;
int temperature;
long gyro_x_cal, gyro_y_cal, gyro_z_cal;
long loop_timer;
int lcd_loop_counter;
float angle_pitch, angle_roll;
int angle_pitch_buffer, angle_roll_buffer;
boolean set_gyro_angles;
float angle_roll_acc, angle_pitch_acc;
float angle_pitch_output, angle_roll_output;

//Initialize the LCD library
LiquidCrystal_I2C lcd(0x3F, 16, 2);

void setup() {
  Wire.begin(sda, scl);                                                        //Start I2C as master
  Serial.begin(115200);                                               //Use only for debugging
  pinMode(LED_BUILTIN, OUTPUT);                                                 //Set output LED_BUILTIN (LED) as output

  setup_mpu_6050_registers();                                          //Setup the registers of the MPU-6050 (500dfs / +/-8g) and start the gyro

  digitalWrite(LED_BUILTIN, HIGH);                                              //Set digital output LED_BUILTIN high to indicate startup

  lcd.init();                                                       //Initialize the LCD
  lcd.backlight();                                                     //Activate backlight
  lcd.clear();                                                         //Clear the LCD

  lcd.setCursor(0, 0);                                                 //Set the LCD cursor to position to position 0,0
  lcd.print("  MPU-6050 IMU");                                         //Print text to screen
  lcd.setCursor(0, 1);                                                 //Set the LCD cursor to position to position 0,1
[... 122 lines cut ...]
  if (lcd_loop_counter == 10)lcd.print(abs(angle_roll_buffer) / 1000); //Print first number
  if (lcd_loop_counter == 11)lcd.print((abs(angle_roll_buffer) / 100) % 10); //Print second number
  if (lcd_loop_counter == 12)lcd.print((abs(angle_roll_buffer) / 10) % 10); //Print third number
  if (lcd_loop_counter == 13)lcd.print(".");                           //Print decimal point
  if (lcd_loop_counter == 14)lcd.print(abs(angle_roll_buffer) % 10);   //Print decimal number
}

void setup_mpu_6050_registers() {
  //Activate the MPU-6050
  Wire.beginTransmission(0x68);                                        //Start communicating with the MPU-6050
  Wire.write(0x6B);                                                    //Send the requested starting register
  Wire.write(0x00);                                                    //Set the requested starting register
  Wire.endTransmission();                                              //End the transmission
  //Configure the accelerometer (+/-8g)
  Wire.beginTransmission(0x68);                                        //Start communicating with the MPU-6050
  Wire.write(0x1C);                                                    //Send the requested starting register
  Wire.write(0x10);                                                    //Set the requested starting register
  Wire.endTransmission();                                              //End the transmission
  //Configure the gyro (500dps full scale)
  Wire.beginTransmission(0x68);                                        //Start communicating with the MPU-6050
  Wire.write(0x1B);                                                    //Send the requested starting register
  Wire.write(0x08);                                                    //Set the requested starting register
  Wire.endTransmission();                                              //End the transmission
}
```

sanrays10, 2020-05-02:
@ahmadSum1 I have the same issue as you...did you manage to fix it?