esp8266/Arduino #888: WiFi reconnect every 5 minutes of UDP listening

locchi93, 2015-10-14:
I have some problems with my program. It's a simple UDP server listening on port 8888 for incoming data.
All runs great, I can send data to it and I get the reply back.
The problem is that if it keep listening for 5 minutes, without sending data to it, i get this from the 
serial wifi debug
```
rm 0
pm close 7 0 0/303779455
reconnect
f -240, scandone
no Casa 2.4 found, reconnect after 1s
reconnect
f 0, scandone
add 0
aid 3
pm open phy_2,type:2 0 0
cnt 

connected with Casa 2.4, channel 1
dhcp client start...
```
I don't know if the problem is related to the settings of my router... or something else...
After this "reconnect," i'm unable to comunicate with the device through UDP, but I can see it running from the serial monitor.
I constantly check the heap and remains constant to "30072".

Links2004, 2015-10-14:
you need to reopen the UDP socket after a Wifi disconnect.
a simple way is:
```cpp
if(!udp) { 
    udp.begin(1234);
}
```
in the loop.

locchi93, 2015-10-14:
Ok... but this in not the main problem... The problem is the disconnection itself. What causes the disconnection?

Links2004, 2015-10-14:
I have seen something like this only with a very old router.
there I was able to increase the time between the disconnects by changing the beacon interval,
but I wasn't able to fix it.

locchi93, 2015-10-14:
I have an Asus n55u... I will try to play with the beacon interval

Links2004, 2015-10-15:
have you a longer time where you not call delay in you code?

locchi93, 2015-10-15:
This is the loop, it's really basic. With the "yield()" call, the wifi should operate properly:
```
void loop() {
  noBytes = Udp.parsePacket();
  if (noBytes) {
    Udp.read(packetBuffer,noBytes);
    Serial.println("riceived");
    handleMessage(packetBuffer[0],noBytes);
  }
  yield();
}
```

Links2004, 2015-10-15:
at the end off the loop "yield" is done automatically ;)
in this part of the code i not see any problems.
have you the same problem when the ```loop``` is empty?

#### The most possible problems are:
##### Bad WiFi signal / WiFi frequency has interference 
- write out ```WiFi.RSSI();```
- change channel of AP

##### some code needs to many time (SDK lost connection)
- check all for, while, .... in code to be sure they not run to long
  
##### SDK is incompatible with some packages from router
- if even code the only connect to WiFi shows this problem
and its sure that the signal from ESP to AP and from AP to ESP is stable.
- try an other Router

locchi93, 2015-10-17:
I think I've solved the problem some how...
I have changed some configuration on my router (asus n55u), now the connection seems stable!
I have enabled "Snooping IGMP" and this seems the key to make it work... but why?

Links2004, 2015-10-17:
if i remember right IGMP is used to build up the multicast routing.
may if it not enabled the ESP get the wrong packages or non packages and then disconnect the WiFi.

thorhs, 2015-10-18:
I just ran across this myself.  I have one ESP in softAP mode, and another one in station mode.  The first one is sending 20pps UDP traffic to the second one (yes that would be DMX-ish data).  After exactly 5 minutes the station goes 'offline'.  The code is really simple, so there should be no surprises there.

WiFi connection
```c++
void connect() {
  udp.stop();

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    blink();
  }
  udp.begin(7654);
}
```

The loop:
```c++
void loop() {
  if (WiFi.status() == WL_CONNECTED) {
    if (!udp)
      udp.begin(7654);

    int packetSize = udp.parsePacket();
    if (packetSize > 0) {
      int len = udp.read(buffer, 128);
      buffer[len] = 0;

      for (int i = START_ID; i <= END_ID; i++) {
        digitalWrite(pins[i - START_ID], (buffer[i] > 0) ? LOW : HIGH);
      }
      blink();
    }
  } else {
    connect();
  }
}
```
I added the WL_CONNECTED and connect() call to work around the issues.  Without this it would just hang.

The AP is working like a champ, I have not seen any issues with that one that I can think of.

This is running the latest 'official' arduino esp8266 code, SDK 1.2.0.  I have seen other threads where upgrading the SDK to 1.3 or 1.4 may solve issues, anyone know how to do that?  Just copy in the new sdk and change boards.txt to use the new names?

Links2004, 2015-10-18:
the latest git version uses SDK 1.3.0_15_08_10_p1

thorhs, 2015-10-18:
I'll try to grab that and see if that changes anything.  Thanks!

thorhs, 2015-10-18:
Would it be sufficient to use the Statging version, or would I need a git clone and build?

Links2004, 2015-10-18:
a ```git clone``` and then run build/build_board_manager_package.sh shut do it.
you may can use ```--depth``` if you not need the full git history.

thorhs, 2015-10-18:
Hmmm, same thing with 1.3.0 (pulled and generated board manager this morning).  I get just over 5 minutes (6018-6019 packets sent at 20pps, to be exact).  If I wanted to try 1.4.0 (I have it downloaded), would it be more than just replacing the libs and ld?

thorhs, 2015-10-18:
Also, is there some status that I could check that might indicate why it got disconnected?  I doubt there is log, but some status flags I could print out?

Links2004, 2015-10-18:
yes 1.4.0 has some changes with task and idle handling @igrr has done some first test but there is no really usable version now.
the SDK only gives a disconnect event and a reason as uint8_t to meaning is may be linket to this enum:
https://github.com/esp8266/Arduino/blob/cabb450fc13eefaf35a78b9a911d59efd548b81a/hardware/esp8266com/esp8266/tools/sdk/include/user_interface.h#L346
but you not find a real description in the SDK docs.

https://github.com/esp8266/Arduino/blob/cabb450fc13eefaf35a78b9a911d59efd548b81a/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp#L821

https://github.com/esp8266/Arduino/blob/cabb450fc13eefaf35a78b9a911d59efd548b81a/hardware/esp8266com/esp8266/tools/sdk/include/user_interface.h#L386

closed 2016-01-14 by locchi93