esp8266/Arduino #889: Mqtt and AWS

sabas1080, 2015-10-14:
Does anyone to use AWS with ESP8266 and MQTT ?

It can? or single API?

Thanks

DKfraDK, 2015-10-16:
It sure can.
If you have a running MQTT broker eg. Mosquitto on an AWS instance, you can use the MQTT client found here https://github.com/Imroy/pubsubclient to connect to it.

sabas1080, 2015-10-16:
I 'm trying to use the library, Do you will have some example of how to implement it with keys and certificate? Thanks

DKfraDK, 2015-10-19:
Unfortunately I don't. I haven't used it with keys and certificate (I actually don't know if you can use a certificate in PubSubClient, or if there is only username/password protection).
You can see the examples provided in the repo for examples using authentication.

tim-in-oakton, 2015-10-22:
If you are trying to use the new AWS IoT service, then the answer is "not yet".  This service requires HTTPS endpoints, which I don't think have been completely implemented on the 8266 yet.  I plan to take a swing at this shortly (< 1 mo), but I don't think this has been figured out yet.

Regarding running Mosquito or another MQTT broker on an EC2 instance - yeah, this works fine.

grahamehorner, 2015-10-23:
@tim-in-oakton SSL support is present in the esp8266 sdk v1.4 but their are definitely some issues currently

2dom, 2015-12-07:
Any progress on on connecting to the AWS MQTT directly?

tim-in-oakton, 2015-12-07:
Not yet, I think a new TLS lib will need to be ported (mbed may be an option) - The AWS IoT service employs TLS 1.2 and mutual auth for security, this is not currently supported in in the sdk (as I understand it)-

2dom, 2015-12-08:
Shame....thanks for the info!

svdgraaf, 2015-12-09:
You should be able to use the AWS iot REST interfaces, although I see there is also no sha256 available, which we would need for signing the requests: http://docs.aws.amazon.com/AWSECommerceService/latest/DG/rest-signature.html

The rest interface still is accepting tls 1.0 and 1.1, which are currently supported by the WifiClientSecure

tim-in-oakton, 2015-12-09:
I published a project using AWS V4 signing (over http) for an ESP8266 kegerator - https://github.com/tim-in-oakton/ESP8266Kegerator  - it is a little hard coded around using SNS for educational purposes, AWS has published a richer (&larger) interface SDK as well.

We can totally use AWS REST interfaces, but the (very cool) MQTT and state abstraction of the IoT service do require TLS 1.2 and mutual auth-

2dom, 2015-12-19:
Yeah...i am running the AWS SDK on the ESP ATM...works great... But MQTT would be much more elegant....

svdgraaf, 2015-12-19:
Can you point me towards that repo somewhere? I'm having a hard time finding a decent one :)

2dom, 2015-12-21:
https://github.com/awslabs/aws-sdk-arduino

2dom, 2015-12-21:
I am using it wit dynamodb....

svdgraaf, 2015-12-21:
Thanks, I looked at that one, and it seems to work fine with dynamo directly, but the api calls don't support the aws iot rest interface yet (the library assumes the POST is to the endpoint host directly, not a sub url on that host), I'm working on making the changes and creating a PR, would be awesome if we could at least get that part of the aws iot stuff working 'easily'.

Oh, and api-gateway from aws is also not supported in this way (same reason: sub urls don't work), you could setup an api-gateway with just a root url, but that would just be silly :P

svdgraaf, 2015-12-30:
I ended up forking the awslabs repo, it wasn't up to date anymore, and the last update was from a couple of months ago.

As I wanted to use the IOT service (the REST part anway, since mqtt over tls1.2 is out of the question for now), I needed to implement the `AWS v4` signature. I now have something that works, there are still some memory leaks to fix. This was my very first c++ library/project so be gentle with any comments ;)

@tim-in-oakton Your repo was a great help! I reused some parts of it, but decided to implement the aws signature seperately. The ntp stuff is great, but I ended up just creating a borked http request, and using the datetime response from that (that will probably save you some extra resources as well).

You can find the fork here: https://github.com/svdgraaf/aws-sdk-arduino, and a basic [example](https://github.com/svdgraaf/aws-sdk-arduino#esp8266) is in the (updated) readme.

I'll cleanup the code and create a PR for the `awslabs` team, but I'm not sure if they will respond.

The IOT stuff is pretty cool, you can trigger for example lambda on any income shadow changes, and do stuff with that data, I'm currently tracking the ambient temperature in my attic:

[image]

odelot, 2016-03-24:
@svdgraaf using your fork as a starting point, I've managed to write a lib to use MQTT over Websockets on AWS IoT plataform. The lib need some improvements, but can be used as a starting point. https://github.com/odelot/aws-mqtt-websockets

thank you for sharing your work ;-)

svdgraaf, 2016-03-24:
@odelot that's so AWESOME :D Can't wait to start using it :) I will check it out somewhere this or next week :)

tuanalan, 2017-02-15:
@odelot -thanks for great work! I'm new to AWS IOT so please bear with me.
Where in AWS console can I find my " iam-key" and "iam-secret-key"?  Thanks for prompt rely.

char aws_key[]         = "your-iam-key";
char aws_secret[]      = "your-iam-secret-key";

odelot, 2017-02-15:
@tuanalan [](http://docs.aws.amazon.com/general/latest/gr/aws-sec-cred-types.html#access-keys-and-secret-access-keys)

closed 2017-02-15 by sabas1080

marcoavaccaro, 2020-10-27:
Hello guys, i'm trying to connect to AWS wich request:

wifiClient.loadCACert(ca);
wifiClient.loadCertificate(client_cert);
wifiClient.loadPrivateKey(client_key);

Via wifi i can do it with no problems. But the ethernet library present on arduino do not have these 3 classes above.

Someone know how to connect to AWS via ethernet shield using ESP32?

devyte, 2020-10-27:
This is the esp8266 repo, nothing to do with esp32

marcoavaccaro, 2020-10-27:
> This is the esp8266 repo, nothing to do with esp32

Sorry man, but many libs between they are shared.