esp8266/Arduino #890: package_esp8266com_index.json incorrect URL specified for ESPtool

jkennard, 2015-10-14:
The json file for loading the ESP8266 board is incorrect (http://arduino.esp8266.com/stable/package_esp8266com_index.json). The url for IGRR's ESPtool is https://github.com/igrr/esptool-ck/releases/download/0.4.5/esptool-0.4.5-win32.zip but it should be https://github.com/igrr/esptool-ck/releases/tag/0.4.5/esptool-0.4.5-win32.zip

lrmoreno007, 2015-10-15:
Read again it...
https://github.com/esp8266/Arduino/blob/esp8266/README.md

pgollor, 2015-10-16:
I think https://github.com/igrr/esptool-ck/releases/download/0.4.5/esptool-0.4.5-win32.zip is correct.

igrr, 2015-10-17:
releases/download/<tag>/<file> works for me.
@jkennard could you please elaborate why do you think this is the wrong one?

closed 2015-11-05 by igrr