esp8266/Arduino #891: ESP_COM OPEN FAILED

Mohakjuneja, 2015-10-15:
I'm using a PL2303 USB to TTL converter with my ESP8266 and Arduino Uno to get it working. 
I connect the 3.3V power supply from the TTL converter to the ESP VCC and CH_PD and connect RX-RX and TX-TX from ESP to the Arduino. The GND pin from the TTL is connected to the GND on ESP. 
When uploading an empty sketch, I get an error - espcom_open failed.
I've tried everything to make it work yet I don't understand what's wrong. Somebody help!

drmpf, 2015-10-15:
I use the programming method described in my http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html
Works reliably for me.
When I occasionally get a programming error, I find exiting the IDE, removing the USB connection and then restarting solves it.

lrmoreno007, 2015-10-15:
Connect TX(ESP) with RX(PL2303), and RX(ESP) with TX(PL2303), and upload the program with "Arduino IDE" to the ESP in the COM port asigned to the PL2303.

After programming, connect TX(ESP) with RX(Arduino), and RX(ESP) with TX(arduino), and communicate between both, if it's what you want. But caution Arduino use 5V level and you can burn the ESP.

You must see other stuff as GPIO0, CH_PD, etc to set the ESP in programming mode or run mode. Read this: https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/boards.md

igrr, 2015-10-20:
@Mohakjuneja which OS are you using? In case it's OS X, PL2303 is not currently supported ([see docs](https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/boards.md#serial-adapter)).
Please verify your wiring by checking if you are able to flash anything using esptool.py.

closed 2015-11-15 by igrr

jopicardoso, 2016-09-27:
This business is kinda frustrating sometimes, and that's putting mildly. Three errors seem to crop up every  time I try to get the ESP8266 to work: espcomm_sync failed, espcomm_open failed and espcomm_upload_mem failed. I'm using ESP-01 with FT232RL USB to Serial. I get the same error messages when using the ESP8266-012 with the same FT232RL.
To make it more puzzling: there have been times when I get a dozen or more errors and warnings on the screen but I **can** use the AT commands despite. So, what is going on? Is there magic in this stuff? Or is it one of the Trump's tricks that come up only to annoy you? :(

drmpf, 2016-09-27:
I regularly get espcomm_sync failed and fix it by
i) power cycle the ESP8266 with GPIO0 grounded (clean power application, see below)
ii) if i) does not fix it unplug the USB cable from the computer wait  few secs and plug it back in
iii) if ii) does not fix it, uplug USB cable from PC, close Arduino IDE, open Arduino IDE, plug USB cable back in.

Noticed I was getting this a lot if I did not I did not cleanly apply power to the ESP. (i.e. ESP flickered as I plugging the the header pins)  I use the SparkFun USB cable as shown here http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html

jopicardoso, 2016-09-28:
Thanks for the tips, I will try them.
'-John-

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

jopicardoso, 2016-09-29:
dr. mpf :)

Thanks for your help. Indeed it was "magic". Your first option (i) did 
it. It all works now, although I have to repeat the crazy manoeuvre 
every time I load some code. Does anybody know why?
I'm a happy camper now. Thanks.

liderbug, 2017-09-06:
OK, I have a Makerfocus esp8266 board, uUSB port, rst & flash buttons (but no model num). It's connected to my Pi3.  I have Arduino IDE etc. installed and can detect and start to Upload (blink) and >splat<
[code]
/home/pi/.arduino15/packages/esp8266/tools/esptool/0.4.8/esptool -vv -cd nodemcu -cb 115200 -cp /dev/ttyS0 -ca 0x00000 -cf /tmp/arduino_build_413565/Blink.ino.bin 
[/code]
produces:
[code]
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from /dev/ttyUSB0 to /dev/ttyS0
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/ttyS0 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	set timeout failed
	done
error: failed sending 0xC0
	espcomm_send_command: sending command header
error: failed sending 8 bytes
...
[/code]

Connecting with 'screen' sits there until I type AT and it is the same as if I do the cat /dev/ttyS0 it looks like it's dumping all 4M of the 8266. I tried using the usb port to no avail so I pulled out my ribbon cable, breakout board and jumped 3v3, gnd, tx,rx. That's when the cat/screen started working. I found a post - pull down gpio0 - no change. I at Whitt's End. Oh, I can ping 192.168.4.1 - problem is my net is 192.168.ZERO.x.
Help please.
Thanks