esp8266/Arduino #892: WIFI unstable and reset periodically

nik600, 2015-10-15:
Dear all

i'm experiencing some stability issue of the module, this is my environment:

ESP8266 - ESP-01

i've created a very simple webserver with 2 features:

- view che time
- set the time

the timing features are based on the internal clock of the module, no RTC module or NTP sync are active, just to check how precise is the internal clock.

I've noticed that after some minutes of work in WIFI_STA mode, the module automatically resets and the http connection is closed, after an hardware reset, or some minutes waiting, the module is up and running again (but obviously, lost the internal settings).

I've also added a counter in the setup() method to trace in the EEPROM the number of restarts after a time set, and in 8 hours the module restarts in average 5/6 times.

any suggestion?

 here my code:

```c
#include <ESP8266WiFi.h>
#include <WiFiClient.h> 
#include <ESP8266WebServer.h>
#include <EEPROM.h>
#include <Time.h>

/* Set these to your desired credentials. */
const char *ssid = "ESPap";

ESP8266WebServer server(80);

/* Just a little test message.  Go to http://192.168.4.1 in a web browser
 * connected to this access point to see it.
 */
void handleRoot() {
  int valore = now();

  int hr=hour();
  int minutes=minute();
  int sec=second();
  int day=day();
  int month=month();
  int yr=year();
  
  int counter = EEPROM.read(1);
  
  String message="number of reset=";
  message+=counter;
  message+=" corrent time = ";
  message+=day;
  message+="/";
  message+=month;
  message+="/";
  message+=yr;
  message+=" ";
  message+=hr;
  message+=":";
  message+=minutes;
  message+=":";
  message+=sec;
  

  server.send(200, "text/html", message);
  
}

void handleSetup() {

  String hr = server.arg("hr");
  String minutes = server.arg("min");
  String sec = server.arg("sec");
  String day = server.arg("day");
  String month = server.arg("month");
  String yr = server.arg("yr");
  
  setTime(hr.toInt(),minutes.toInt(),sec.toInt(),day.toInt(),month.toInt(),yr.toInt());

  String message="time changed succesfully and reset cunter ";
  
  EEPROM.write(1,0);
  EEPROM.commit();

  server.send(200, "text/html", message);
  
}

void setup() {
	delay(1000);
	Serial.begin(9600);
  EEPROM.begin(512);

  int counter = EEPROM.read(1);
  counter++;
  EEPROM.write(1,counter);
  EEPROM.commit();

  WiFi.mode(WIFI_STA);
  WiFi.softAP(ssid);    

  server.on("/", handleRoot);
  server.on("/setup", handleSetup);
	server.begin();
  
}

void loop() {
	server.handleClient();
}
```

devyte, 2017-10-20:
@nik600 I have something not too different running in my app, and I don't have stability issues. 
Closing as can't reproduce.
If there is still a problem with latest git, please open a new issue and fill in the requested info.

closed 2017-10-20 by devyte

rsegecin, 2017-11-16:
I'm having a similar problem but worse because it's quite frequent and it doesn't appear to be the WDT, it seems that reboots clearing the terminal but it doesn't show any message. It's most common when the client disconnects and reconnects multiple times. I made an [Android App](https://github.com/rsegecin/SwitchNetwork) that switches the wireless connection to reconnect with ESP module configured with the SSID same as in the code below. 

Here's the code:

	#include <ESP8266WiFi.h>
	#include <WiFiClient.h> 
	#include <ESP8266WebServer.h>

	#define LED_PIN 2

	const char ssid[6] = "ESPap";

	ESP8266WebServer server(80);
	int stateLED = LOW;

	char lBuffer[200];

	const char strLEDState[] PROGMEM = "LED on state: %u";
	const char webLEDState[] PROGMEM = "{\"ledstate\": \"%u\"}";

	void setup()
	{
		delay(1000);

		pinMode(LED_PIN, OUTPUT);
		digitalWrite(LED_PIN, HIGH);

		Serial.begin(115200);
		Serial.println();
		Serial.println();
		Serial.println("Configuring access point...");

		IPAddress apIP(192, 168, 0, 1);
		WiFi.mode(WIFI_AP_STA);
		WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
		WiFi.softAP(ssid);

		IPAddress myIP = WiFi.softAPIP();
		Serial.print("Server address: ");
		Serial.println(myIP);
		server.on("/", handleRoot);
		server.begin();
		Serial.println("HTTP server started");
	}

	void loop()
	{
		server.handleClient();
		yield();
	}

	void handleRoot()
	{
		if (stateLED)
			stateLED = LOW;
		else
			stateLED = HIGH;

		digitalWrite(LED_PIN, stateLED);

		sprintf_P(lBuffer, webLEDState, stateLED);
		server.send(200, "application/vnd.api+json", lBuffer);

		sprintf_P(lBuffer, strLEDState, stateLED);
		Serial.println(lBuffer);
	}