esp8266/Arduino #893: ESP8266 not enabled in AtmelStudio7

grahamehorner, 2015-10-15:
I have installed the board/plugin for the Arduino IDE and I'm able to use the Ardunio as expected; however the ESP8266 board/plugin is not shown in the AtmelStudio 7 like the other board/plugins

asetyde, 2015-11-04:
Has say in VisualMicro forum page as site , Atmel doesn't want other platform , also i 've use but at now Visual studio 2015 community is best choice . (at now tested only with this stable version)

grahamehorner, 2015-11-04:
@asetyde sorry, but weather Atmel want it or not given VisualMicro give you the option to install the plug-in it should give the same experience as on VisualStudio 2015 (which it doesn't) I have both AtmelStudio 7 (which is base on/powered by VisualStudio) and VisualStudio 2015 Enterprise as far as I can tell the issue looks to be with the board manager which has been logged with VisualMicro; this may also be related to compiler include path issue and/or the location of the esp8266 core package.

If it can't be achieved because of the implementation of AtmelStudio then that is a different matter; it should be reported as a issues to Atmel and/or this AtmelStudio dropped as an option in my opinion.

The big question is why do other 3rd party boards show up in the target list?

ATMEL DEVELOPMENTS are customers the CUSTOMER is always right ;)

asetyde, 2015-11-04:
----> From Visual Micro Forum - sorry friend but you never see :(
Re: Is Atmel Studio 7 Supported?
Reply #5 - Oct 15th, 2015 at 1:46am Mark & Quote Quote  Hi, AS7 is new and the documentation was produced before it was available. 

However there are benefits to both Ide's. Here is a selection of random observations:-

Arduino code is designed to run on different processors and we see a lot of varieties such as the Intel boards that arduino.cc promote and also boards such as the $10 Esp8266 which runs at 80mhz and also has wifi. The Atmel license doesn't permit these boards to be used so from a support perspective it's easier to support Vs.

The intellisense in Visual Studio can often be easier for new users because code is shown as disabled when a conditional #define resolves to false. Vs2015 also shows methods and variables expanded below file names in the solution explorer. 

For certain low level code the Atmel Studio intellisense can perform better because it knows gcc where as Microsoft have their own C++ syntax. This doesn't affect 99% of users and a nice feature of Vs2015 is the finer control of the displayed of intellisense errors so it's not so much of an issue as it was in the past.

Windows telemetry and IoT programs are of interest to many. Vs2015 has a wide range of tools that allows these programs to be built. If users want to install a single Ide then Vs2015 makes sense.

AS7 has a few quirks that have the potential to confuse users such as automatically adding a main.cpp and it's own configuration sources to new or upgraded projects. The main.cpp is especially a problem because it will break sketches and prevent them from working when compiled with the arduino ide. Visual Micro As7 monitors for this issue and prompts to auto-fix but it's not ideal.

There are also a lot of changes happening within the Arduino eco system so the software is constantly being updated to cater for these changes. There is a problem with the Atmel gallery at the moment that prevents extensions from being uploaded unless they use an older Microsoft format. When people download from the gallery the automatic update notification system of the gallery is activated for the extension. The auto update system is important otherwise the forum fills up with questions from people who have upgraded their arduino ide cores but not Visual Micro.

It really does depend on what is important to you. The download link for as7 below was out of date. I have changed the link to point to the downloads page where the latest As7 extension can be found. Atmel are certainly investing heavily and have a great team working on AS7 and if you want to explore the avr,sam or samd architectures in depth then it is a must.

Personally I have more of a relationship and assistance from Atmel than Microsoft so I am in their camp but most of Visual Micro is free and time becomes short so support comes first. Until this week the release of As7 could fail to start after installing/uninstalling extensions. It was a Microsoft problem that Atmel have "worked around" but it would have reflected badly on Visual Micro. I expect we will see a few releases of As7 over the next few months and will monitor its progress.
Back to top	« Last Edit: Oct 15th, 2015 at 2:00am by Visual Micro »

asetyde, 2015-12-01:
Please close this issue , all go

closed 2015-12-01 by Links2004