esp8266/Arduino #894: issue with SPIFFS 64K image creation, Size limitation or bug ?

luc-github, 2015-10-16:
Hi when I try to use the Generic Module with 512K/64K SPIFFS for my ESP01 I just have 
SPIFFS Create Failed! with no explanation

[image]

I noticed size on disk versus real size is 76K so I have deleted until get 56K and still fail even under 64K i all cases
[image]
but if I delete just the biggest file (27K) it is ok
[image]
if I keep biggest one but delete several files to be close to 44K it is also OK
[image]

I have no issue using ESP01 1M/512K SPIFFS for image creation but module is not working 
 with nodemcu 1.0 any SPIFFS size no issue at all

Is there any limitation for 64K SPIFFS that we cannot use more than half of the size ? or it is a bug ?

Thanks

luc-github, 2015-10-16:
the best I was able to do was 41.3K real with 52K on disk - 

[image]

pgollor, 2015-10-16:
I added some debug outputs to the `mkspiffs` tool and tried it with your dir from here https://github.com/luc-github/ESP8266/tree/master/esp8266/data.

The function `SPIFFS_write` fails for the following files:
 * `/printer.tpl`
 * `/header.inc`
 * `/home.tpl`

But image creation with this three files only is working.

I think it is a general problem from the SPIFFS file system. Mybe there are not enough blocks left or so.

Here is the fully output from my debug `mkspiffs` version:
```
mkspiffs -d 1 -c ./data/ data.bin 
Debug output enabled
/config_ap.tpl
file opend
file size: 3419
file closed and writed sucessfully

/css.inc
file opend
file size: 4983
file closed and writed sucessfully

/config_sta.tpl
file opend
file size: 3514
file closed and writed sucessfully

/settings.tpl
file opend
file size: 1685
file closed and writed sucessfully

/restart.tpl
file opend
file size: 438
file closed and writed sucessfully

/favicon.ico
file opend
file size: 1150
file closed and writed sucessfully

/system.tpl
file opend
file size: 1463
file closed and writed sucessfully

/404.tpl
file opend
file size: 518
file closed and writed sucessfully

/printer.tpl
file opend
file size: 25871
SPIFFS_write error with -1!
data left: 1295
error adding file!

/footer.inc
file opend
file size: 31
file closed and writed sucessfully

/header.inc
file opend
file size: 985
SPIFFS_write error with -1!
data left: 985
error adding file!

/home.tpl
file opend
file size: 3342
SPIFFS_write error with -1!
data left: 3342
error adding file!
```

pgollor, 2015-10-16:
I checked the error code form `SPIFFS_write` and it is: `-1001` which means `SPIFFS_ERR_FULL`
I think the best way is to add an issue here: https://github.com/pellepl/spiffs

luc-github, 2015-10-16:
actually doing trial/error I got a 64K image working by removing some files and merging the content of some to others https://github.com/luc-github/ESP8266/tree/master/esp8266/data%20-%20for%2064K%20SPIFFS
but this limit content to 41.3K real  / 52K on disk for 7 files which is weird

@pgollor thanks, I have checked github and it seems latest version it is 0.3.3 when it seems ESP package has mkspiffs 0.1.1 I do not know  if versions are related

pgollor, 2015-10-16:
mkspiffs use spiffs 0.2 https://github.com/igrr/mkspiffs/commit/862a237150d2d70cb5197ba7ad3fd7bf1216827d

luc-github, 2015-10-16:
Thanks, I will install MinGW and have a look

pgollor, 2015-10-16:
You can check the free space with `mkspiffs -i data.bin`. This function prints the maximum available size. For 64kb it is 52961.

pgollor, 2015-10-16:
Spiffs version 0.3.3 does not make a difference. You can try it if you want: https://github.com/igrr/mkspiffs/pull/10

luc-github, 2015-10-17:
Yes I did - thanks a lot

luc-github, 2015-10-17:
So seems normal to have 52.9K usable for 64K SPIFFS - I will work with this limitation

Thanks for the help

closed 2015-10-17 by luc-github