esp8266/Arduino #896: esp smartconfig (esp touch) does not work with WiFi.mode(WIFI_AP_STA);

tzapu, 2015-10-16:
i've just had a play with esp touch and I can not make it work in AP_STA mode at all.
seems to work most of the time in WIFI_STA mode

igrr, 2015-10-18:
Does it work with Espressif's SDK (without Arduino addon)?
Our smartconfig wrapper is quite thin, there isn't anything which depends on WiFi mode there as far as i can tell.

tzapu, 2015-10-19:
unfortunately i don't have the knowhow to test that. do they have an issues forum?

cheers and feel free to close this if it does not pertain to the arduino addon

romuye123, 2015-11-25:
esp smartconfig was working for me earlier in AP_STA mode.
but there was some library issue, so i installed the arduino ide freshly and now it only works with STA mode.

for AP_STA mode it crashes and gives exception.

linanw, 2016-06-09:
It works for me only on WIFI_STA mode, never worked on WIFI_AP_STA mode.

romuye123, 2016-06-21:
@igrr : any updates on this ? I want to run both AP and STA mode. So user have option to connect with device.

igrr, 2016-06-21:
Quoting SDK documentation (API guide, "SmartConfig APIs" section):
```
Note:
• This API can only be called in station mode.
```
So i don't think this will work in AP or AP_STA mode.

devyte, 2017-10-20:
Per [Espressif api reference for sdk 2.1.0](https://github.com/espressif/ESP8266_NONOS_SDK/releases/tag/v2.1.0), the previous comment is correct.
Closing.

closed 2017-10-20 by devyte