esp8266/Arduino #897: How to know the paramenters of the board

DiegoDVG, 2015-10-16:
Hi, congratulation for the projet!!

my question is how or where i can search the parameters memory, frecuency and other of the bards

fr example: 

http://www.banggood.com/es/Upgraded-Version-1M-Flash-ESP8266-ESP-01-WIFI-Transceiver-Wireless-Module-p-979509.html

igrr, 2015-10-17:
@DiegoDVG the product page mentions that flash size is 512k.

Frequency is not really a parameter of the board. All ESP8266 boards sold nowadays have a 26 MHz oscillator. CPU frequency is always 80 MHz (obtained using a PLL), unless you want to overclock it at 160 MHz. This may be done using a sub-menu in Arduino Tools menu.

penfold42, 2015-10-18:
And if you're unsure of the flash size, look up the data sheet or write a small sketch to print the flash chip ID and then look it up online

Links2004, 2015-10-18:
this example gives you all info's about the flash, and also checks for valid config in the IDE:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/esp8266/examples/CheckFlashConfig/CheckFlashConfig.ino

closed 2015-10-18 by igrr