esp8266/Arduino #898: Servo library can not work

szpapas, 2015-10-18:
git version:  2015-10-18:
From https://github.com/esp8266/Arduino
   d8f57cb..bda5cff  esp8266    -> origin/esp8266
Updating d8f57cb..bda5cff

==== Load Servo example ===
/* Sweep
 by BARRAGAN <http://barraganstudio.com> 
 This example code is in the public domain.

 modified 28 May 2015
 by Michael C. Miller
 modified 8 Nov 2013
 by Scott Fitzgerald

 http://arduino.cc/en/Tutorial/Sweep
*/ 

#include <Servo.h> 
 
Servo myservo;  // create servo object to control a servo 
                // twelve servo objects can be created on most boards
 

void setup() 
{ 
  Serial.begin(115200);
  myservo.attach(4);  // attaches the servo on GIO2 to the servo object 
} 
 
void loop() 
{ 
  int pos;
  Serial.println("begin rotate....");
  
  for(pos = 0; pos <= 180; pos += 1) // goes from 0 degrees to 180 degrees 
  {                                  // in steps of 1 degree 
    myservo.write(pos);              // tell servo to go to position in variable 'pos' 
    delay(15);                       // waits 15ms for the servo to reach the position 
  } 
  for(pos = 180; pos>=0; pos-=1)     // goes from 180 degrees to 0 degrees 
  {                                
    myservo.write(pos);              // tell servo to go to position in variable 'pos' 
    delay(15);                       // waits 15ms for the servo to reach the position 
  } 
} 

==== build and running on ESP12 ====
使用库 Servo 在文件夹： /Applications/Arduino.1.6.6.app/Contents/Java/hardware/esp8266com/esp8266/libraries/Servo 

/Applications/Arduino.1.6.6.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Applications/Arduino.1.6.6.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/include -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/Applications/Arduino.1.6.6.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266 -I/Applications/Arduino.1.6.6.app/Contents/Java/hardware/esp8266com/esp8266/variants/generic -I/Applications/Arduino.1.6.6.app/Contents/Java/hardware/esp8266com/esp8266/libraries/Servo/src /var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/Sweep.cpp -o /var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/Sweep.cpp.o 
使用已经编译的文件：/var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/Servo/esp8266/Servo.cpp.o
/Applications/Arduino.1.6.6.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Applications/Arduino.1.6.6.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/include -c -g -x assembler-with-cpp -MMD -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/Applications/Arduino.1.6.6.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266 -I/Applications/Arduino.1.6.6.app/Contents/Java/hardware/esp8266com/esp8266/variants/generic /Applications/Arduino.1.6.6.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/cont.S -o /var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/cont.S.o 
使用已经编译的文件：/var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/cont_util.c.o
使用已经编译的文件：/var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/core_esp8266_eboot_command.c.o
使用已经编译的文件：/var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/core_esp8266_flash_utils.c.o
使用已经编译的文件：/var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/core_esp8266_i2s.c.o
使用已经编译的文件：/var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/core_esp8266_noniso.c.o
使用已经编译的文件：/var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/core_esp8266_phy.c.o
使用已经编译的文件：/var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/core_esp8266_postmortem.c.o
使用已经编译的文件：/var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/core_esp8266_si2c.c.o
使用已经编译的文件：/var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/core_esp8266_timer.c.o
使用已经编译的文件：/var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/core_esp8266_wiring.c.o
使用已经编译的文件：/var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/core_esp8266_wiring_analog.c.o
使用已经编译的文件：/var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/core_esp8266_wiring_digital.c.o
使用已经编译的文件：/var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/core_esp8266_wiring_pulse.c.o
使用已经编译的文件：/var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/core_esp8266_wiring_pwm.c.o
使用已经编译的文件：/var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/core_esp8266_wiring_shift.c.o
使用已经编译的文件：/var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/libc_replacements.c.o
使用已经编译的文件：/var/folders/42/w4dvmhl9257dj09078s22xch0000gn/T/build5613857381500289639.tmp/spiffs_cache.c.o
[... 524 lines cut ...]
3fffff70:  3ffeb734 000000f5 3ffe8f70 3ffe8f70  
3fffff80:  00000080 3ffeca48 00000000 3ffee940  
3fffff90:  402097e8 3fffdab0 00000000 3fffdcc0  
3fffffa0:  3ffe8f70 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
begin rotate....

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
begin rotate....

chrisallick, 2015-10-18:
Hey, I was having hte same problem and I solved it by attaching my servo at the END of my setup function.

igrr, 2015-10-18:
@szpapas thanks for reporting this. Please try the fix in e5a5456.

closed 2015-11-30 by igrr