esp8266/Arduino #900: Tone not working

questuk, 2015-10-18:
Hi

Have spent an hour trying to solve this and looked on the internet.

When i add this #include <Tone.h> i get a fatal error.

'libraries\Tone\Tone.cpp:24:27: fatal error: avr/interrupt.h: No such file or directory #include <avr/interrupt.h>'

But I am not sure if the Tone library has been implemented on the ESP8266 ?

Regards

Gary

Links2004, 2015-10-18:
Tone is not implemented for ESP
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/cores/esp8266/Tone.cpp#L46

gorec2005, 2015-10-18:
I test "tone" work ok only if use pwm for it...

dirty func:
void tone(uint8_t _pin, unsigned int frequency, unsigned long duration) {
pwmFreq(frequency);
analogwrite(_pin,500);
delay(duration);
analogwrite(_pin,0);
}

igrr, 2015-10-18:
Same as #582?

alx-1, 2015-10-29:
gorec2005, I get this error when compiling with your tone function : 'pwmFreq' was not declared in this scope'.  Do you have a separate function for it?

gorec2005, 2015-10-30:
don't mention it... i think somebody implement it to the "kernel" of arduino-esp IDE :-) i don't have enought knowlege for this :-(

Links2004, 2015-11-27:
closed will be tracked at #582

closed 2015-11-27 by Links2004