esp8266/Arduino #901: SPIFFS Create Directory API

luc-github, 2015-10-18:
Sorry if it is stupid question but I cannot find the answer
the API reference http://arduino.esp8266.com/versions/1.6.5-1160-gef26c5f/doc/reference.html#open
```
Dir dir = SPIFFS.openDir("/data");
while (dir.next()) {
    Serial.print(dir.fileName());
    File f = dir.openFile("r");
    Serial.println(f.size());
}
```

 it says openDir allow to parse a directory like __/data__ - today I can parse __/__ but how to create a directory lie data ?

igrr, 2015-10-18:
SPIFFS doesn't have directories, it's a flat file system. You can create a
file called /data/name.txt, then when you do openDir("/data"), it will be
listed. Basically openDir function allows you to filter files based on some
prefix.

luc-github, 2015-10-18:
understood, thanks a lot

closed 2015-10-18 by luc-github