esp8266/Arduino #903: esp8266 eeprom read crash

ADJ7WTB, 2015-10-18:
Does any one know why exception 28 crash on esp8266. I write data to eeprom. But when I want to read it. The esp8266 crash. 
My code:
#include <EEPROM.h>

Struct mydata{
  char ssid;
  char pass;};

void setup(){ 
Serial.begin(9600);
EEPROM.begin(4096);
Readeeprom();
}
void Readeeprom (){
const char* said;
const char* pass;

mydata Varcustom;

EEPROM.get(0, Varcustom);
//Exception 28 error begin here
Serial.println(Varcustom.ssid);// does not show this print
}

Any help??

ADJ7WTB, 2015-10-18:
My write to eeprom code look like this

My code:
#include <EEPROM.h>
Struct mydata{
char ssid;
char pass;
};

void setup(){ 
Serial.begin(9600);
EEPROM.begin(4096);
WriteEeprom();
}
void WriteEeprom (){
mydata Varcustom = {"ssid","password"};
EEPROM.put(0, Varcustom);
}

The short simple version

igrr, 2015-10-18:
This code doesn't look like it should work.
If you need to save strings to EEPROM, take a look at this example: https://github.com/tzapu/WiFiManager/blob/master/WiFiManager.cpp#L164-L194

ADJ7WTB, 2015-10-18:
Thanks for reply. 
According to arduino ide document it should work.
I tried converting string to char. Then write it to eeprom. Should work but it gives me the same error. Will look at that code anyways.

igrr, 2015-10-18:
Your writing code is not well-formed because struct fields are `char`s, and you are initializing these fields with `const char*`. This is undefined behaviour in C++.

Reading code doesn't make sense (you are only reading a single character for `ssid` and single character for `pass` field), but it shouldn't crash. I modified your sample a bit (changed baud rate, changed `Struct` to `struct`, added a `Serial.println` to the end of setup function): https://gist.github.com/igrr/cfd5411dccb2c6764e41. It runs without issues on my side.

ADJ7WTB, 2015-10-19:
Thanks for the help. Atleast it's not crashing.

asetyde, 2015-11-04:
I think is not good idea , use "block function" for me is stable and best choice 

void eeprom_read_block(uint8_t * data, uint32_t address, size_t len) {
  int i;
  for (i = 0; i < len; i++) {
    data[i] = EEPROM.read(address + i);
  }
}

void eeprom_write_block(uint8_t * data, uint32_t address, size_t len) {
  int i;
  for (i = 0; i < len; i++) {
    EEPROM.write(address + i, data[i]);
  }
  EEPROM.commit();
}

asetyde, 2015-11-04:
simple and full tested

closed 2015-11-15 by igrr