esp8266/Arduino #908: esp8266 sdk updates

grahamehorner, 2015-10-20:
update to use esp8266 sdk v1.4.0 - esp_iot_sdk_v1.4.0_15_09_18.zip and provide a method of obtaining the sdk used for the build of core libraries.

igrr, 2015-10-20:
There is a branch based on SDK 1.4: https://github.com/igrr/Arduino/tree/sdk-1.4
However currently there are some issues with 1.4 related to AP mode and sleep modes, so I'm not upgrading the core yet.

Regarding the second part, i'm not sure i understand what you mean by 'provide a method of obtaining the sdk used for the build of core libraries'. SDK is in this repository, https://github.com/esp8266/Arduino/tree/esp8266/hardware/esp8266com/esp8266/tools/sdk. You can see which version is used by looking at [version file](https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/tools/sdk/version). If you need to get this information at run time, there is a function in the SDK which returns version number. See SDK docs for details.

igrr, 2015-12-18:
Updated to 1.5 (#1102)

closed 2015-12-18 by igrr