esp8266/Arduino #909: checking encryption of connected AP

chaeplin, 2015-10-20:
I read http://mobile.slashdot.org/story/15/10/20/1618209/tattling-kettles-help-researchers-crack-wifi-networks-in-london
https://www.pentestpartners.com/blog/hacking-kettles-extracting-plain-text-wpa-psks-yes-really/
---------------------------------------------
The brief version:
De-auth kettle from its usual access point. Use aireplay-ng
Create fake AP with same SSID
Kettle joins
Connect to telnet service, authenticate using default PIN of ‘000000’
Enter ‘AT-KEY’
Plaintext WPA PSK is then disclosed
Yes, it’s that easy
-----------------------------------------------

So, I changed my AP's setting from WPA2/AES to OPEN(security disabled) with same SSID.
Then my esp8266 connected to the AP without problem.
(configured with   WiFi.begin(ssid, password);)

How to get the encryption type of the current network ?

Ref: https://www.arduino.cc/en/Reference/WiFiEncryptionType

drmpf, 2015-10-21:
So you are saying that, ESP has the same or similar, problem.  
That is even if you set up a secure ssid, password on your ESP, it will still connect to an insecure (OPEN) network with the same ssid. 
However in the ESP case there is no 'display password' command built in, so not as bad as the kettle, but still not good as you can get access to the ESP device and send it commands.

p.s. When using my Android app, pfodApp, to control WiFi devices, like the ESP, I recommend using the 128bit security option, which adds a secure hash to each msg in both directions to provide end to end security.  Commands are still in plain text but without the correct hash they will not be acted on.

tim-in-oakton, 2015-10-23:
@drmpf - that approach is still subject to a replay attack, so ensuring idempotent commands would be recommended.

drmpf, 2015-10-23:
@tim-in-oakton  Yes ESP could be controlled from the open network so still not good.

My approach (http://www.forward.com.au/pfod/secureChallengeResponse/index.html)
protects against replay attack by adding a hidden message seq no to each msg which is included in the hash.  The message hashes are calculated initializing SipHash with the “secret key” and then adding the challenge to the hash and then adding the message count in bigEndian format, starting from zero and finally adding the message. 
The challenge is different for each and every connection so you cannot replay the whole connection either.

igrr, 2015-11-16:
It appears that there is no SDK API to get the encryption mode of the current connection. Perhaps this needs to be raised to Espressif.

tablatronix, 2016-02-19:
I don't know of this has changed yet, you could scan networks, store BSSIDs and encryptiontypes from the scan,  then recheck a connected BSSID against it.. Of course if BSSID is spoofed... shrug

devyte, 2017-10-20:
There is currently no api available to obtain the encryption type being used by the station. The only aternative I know of at this time is that described in the previous comment: do a scan, and check for authTypes. You could check that all APs with your relevant SSID have the same authType that you need.
If any differs, then it's an error, and you wouldn't connect.
This kind of logic is too complex to be included in the WiFi lib. It better belongs in a Wifi manager, and the requirements for a wifi manager are different from user to user (there is a simple WiFiManager object in the repo, but e.g. it's not enough for my purposes).
Considering that there is nothing that can be done in this repo, I am closing this.

igrr, 2018-02-22:
SDK now supports filtering APs by their AUTH_MODE. This can be used to prevent connection to open APs when passphrase is provided by the application.

The implementation can be as simple as
```c++
    if (passphrase && strlen(passphrase)) {
        conf.threshold.authmode = AUTH_WPA_PSK;
    } else {
        conf.threshold.authmode = AUTH_OPEN;
    }
```
, but this needs to be tested.

artua, 2018-03-20:
@igrr is it still impossible to get current (connected) network encryption type?