esp8266/Arduino #910: setjmp/longjmp appear to be broken

seangoff, 2015-10-21:
I can`t get setjmp/longjmp to work. The following code can be used to reproduce the problem:
#include <setjmp.h>
jmp_buf e;
void setup() {
setjmp(e);
longjmp(e,1);
}
void loop() {
}
when compiled for ESP8266 it throws 
tests.ino:4: undefined reference to `setjmp'
tests.ino:5: undefined reference to `longjmp'
however it compiles just fine for any other Arduino...
What is the problem?

igrr, 2015-10-21:
setjmp/longjmp are normally part of C library, which we currently don't use for certain reasons. Most necessary functions have been added to libc_replacemnts.c in the core, but not setjmp/longjmp.

Also I'm not sure how setjmp/longjmp will interact with context switching which is performed by the core itself (in cont.S). What is your use case for these functions?

igrr, 2015-10-21:
I think setjmp/longjmp will work okay if done within the same context, i.e. if you don't do a jump from WiFi or TCP callback into your Arduino loop() or vice versa.
You can add setjmp/longjmp source to libc_replacements.c. If it works for you, please consider opening a pull request.

malachib, 2016-03-16:
+1 on this issue, I would like setjmp/longjmp availability

sayak-brm, 2016-07-30:
> I would like setjmp/longjmp availability

Me too!

sayak-brm, 2016-07-30:
> You can add setjmp/longjmp source to libc_replacements.c. If it works for you, please consider opening a pull request.

How?

magore, 2016-08-18:
They have a function that does part of this.
A good starting point would be to look at the Arduino Yield code for the esp8266
specifically these functions: cont.S,  cont.h, cont_util.c

lhsantos, 2017-03-22:
any news about this?

d-a-v, 2017-03-22:
under linux it works. I'm using it.

lhsantos, 2017-03-23:
i'm not sure why using it on a different system should make a difference (i believe these functions are implemented in assembly?), but maybe the official distribution doesn't have the latest patches, i'll check it, thanks!

d-a-v, 2017-03-23:
You are right, it is platform independant.
setjmp/longjmp works at least with the latest git version of esp8266/Arduino.

lhsantos, 2017-03-23:
OK, I've confirmed that this works on Linux, but not on Windows. Probably has something to do with the avr-gcc toolchain. I hope this gets fixed on the next release, using a Linux vm for now.

LynXzp, 2017-07-21:
Oh, thanks, under linux it work with latest git version, but not with 2.3.0 release.

rsegecin, 2017-10-11:
I went through this rabbit hole and find out that you can add the reference to **setjmp.h** in windows creating symbolic links to where it resides. Going to directory below and adding the references that are enumerated I ended up missing a reference called **rpc.h** that I could find but it didn't resolve for some reason. If someone go any further please give a shout. I'd like to have **setjmp.h** to enable [Regexp](https://github.com/nickgammon/Regexp) while we don't have _sscanf_ functionality. I have an arduino project that's working with **Regexp.cpp** that uses setjmp so shouldn't be able to be run on esp8266 arduino project as well?

```C:\Users\[UserName]\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266``` 

- ```mklink /J msvc "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\SDK\ScopeCppSDK\VC\include"```
- ```mklink /J windows "C:\Program Files (x86)\Windows Kits\10\Include\10.0.15063.0\um"``` 
- ```mklink /J scopesdk "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\SDK\ScopeCppSDK\SDK\include\shared"```