esp8266/Arduino #914: Any idea, how to join open network with username/password via the http form. Usual in the hotels i.e.

robertgregor, 2015-10-21:


grahamehorner, 2015-10-21:
normally this is done with cookies; the proxy service hotels using look for a session cookie which if present allows traffic to be sent/received as it indicates a valid active session for the client.

torntrousers, 2015-10-22:
I've an example of doing that [here](https://github.com/HarringayMakerSpace/IoT/blob/master/HygrometerOnceADay/HygrometerOnceADay.ino). That is for a BT FON logon in the UK, which wants an HTTPS POST with userid and password. It most likely wouldn't work exactly as-is for a non-BT FON system, but the code might give you something to start with. I think i used the HttpFox Firefox plugin to capture a browser logon to see what it did.

igrr, 2015-10-23:
I honestly feel that esp8266.com forums are a better place for "how do I..." kind of questions. We have a whole subforum devoted to usage of ESP8266 with Arduino there, see http://esp8266.com/arduino.

robertgregor, 2015-10-24:
Yes, you are right, this is only for submitting an issues and bugs. Sorry...

closed 2015-10-24 by robertgregor