esp8266/Arduino #915: NTP using domain name, not IP

adrianrb, 2015-10-21:
Hi all.

How use NTP using DNS?
NTP servers IPs changes sometimes...

Adrian

AndyKirby, 2015-10-21:
Usually only if you are using the pool ntp servers.

It is customary on an internal network to run up the NTP service on
something that doesn't get turned off. Like ummm the DNS server or
something else that is providing your local infrastructure services. In
this case then the IP doesn't change. But that server can if needs be
use the pool NTP servers.

Local infrastructure services are something that should ideally be given
some serious thought for IoT work..... Especially when using lightweight
embedded wifi kit like the ESP. It is a very useful and inexpensive
device but it does have it's limitations and benefits greatly from being
helped along.

For my ESP devices I run a Raspberry Pi as the IoT/wifi gateway and put
all the necessary supporting services for the IoT network on there, so
DNS (Dynamic),DHCP(Updating the DNS so the esp units don't need to do
it),NTP, MQTT etc.

Admittedly if you want to run the ESP standalone you don't have the
benefit of a fully supporting infrastructure.

Kirbs

adrianrb, 2015-10-22:
Yes, I want NTP by name...
Examples:
time.nist.gov, 
 a.st1.ntp.br,
c.st1.ntp.br,
d.st1.ntp.br,
a.ntp.br,
b.ntp.br,
c.ntp.br

Adrian

adrianrb, 2015-10-22:
Perfect, Igrr! Tks!
But, there´s a problem with WiFi.hostByName that prevents to become stable?

igrr, 2015-10-22:
hostByName works fine in stable. It's just the change to the sample sketch has been done after last stable release.

closed 2015-10-22 by igrr