esp8266/Arduino #916: Mysql library

asetyde, 2015-10-21:
I've found this good mysql arduino library , it's work very well and is done for ethernet but also can work with wifi (mysql.h library define).

https://launchpad.net/mysql-arduino

I want adapt it because it's useful but some things not work so well in library  ,
Someone can help me ?

i hope it can be added to our library support

mdilger, 2015-10-22:
I am interested in this library.  Can you tell us what things don't work well?

asetyde, 2015-10-22:
ok first step : i cut ethernet library and spi and uncomment wifi define (change wifi.h with esp....h)
[image]

asetyde, 2015-10-22:
this function give me some error : is in example of library

`/**
 * do_query - execute a query and display results
 *
 * This method demonstrates how to execute a query, get the column
 * names and print them, then read rows printing the values. It
 * is a mirror of the show_results() example in the connector class.
 *
 * You can use this method as a template for writing methods that
 * must iterate over rows from a SELECT and operate on the values read.
 *
 */
void do_query(const char *q) {
  column_names *c; // pointer to column values
  row_values *r;   // pointer to row values

  // First, execute query. If it returns a value pointer,
  // we have a result set to process. If not, we exit.
  if (!my_conn.cmd_query(q)) {
    return;
  }

  // Next, we read the column names and display them.
  //
  // NOTICE: You must *always* read the column names even if
  //         you do not use them. This is so the connector can
  //         read the data out of the buffer. Row data follows the
  //         column data and thus must be read first.

  c = my_conn.get_columns();
  for (int i = 0; i < c->num_fields; i++) {
    Serial.print(c->fields[i]->name);
    if (i < c->num_fields - 1) {
      Serial.print(",");
    }
  }
  Serial.println();

  // Next, we use the get_next_row() iterator and read rows printing
  // the values returned until the get_next_row() returns NULL.

  int num_cols = c->num_fields;
  int rows = 0;
  do {
    r = my_conn.get_next_row();
    if (r) {
      rows++;
      for (int i = 0; i < num_cols; i++) {
        Serial.print(r->values[i]);
        if (i < num_cols - 1) {
          Serial.print(", ");
        }
      }
      Serial.println();

      // Note: we free the row read to free the memory allocated for it.
      // You should do this after you've processed the row.

      my_conn.free_row_buffer();
    }
  } while (r);
  Serial.print(rows);
  Serial.println(" rows in result.");

  // Finally, we are done so we free the column buffers

  my_conn.free_columns_buffer();
}`

asetyde, 2015-10-22:
this important function has some problems :  

/**
 * do_query - execute a query and display results
 *
 * This method demonstrates how to execute a query, get the column
 * names and print them, then read rows printing the values. It
 * is a mirror of the show_results() example in the connector class.
 *
 * You can use this method as a template for writing methods that
 * must iterate over rows from a SELECT and operate on the values read.
 *
 */
void do_query(const char *q) {
  column_names *c; // pointer to column values
  row_values *r;   // pointer to row values

  // First, execute query. If it returns a value pointer,
  // we have a result set to process. If not, we exit.
  if (!my_conn.cmd_query(q)) {
    return;
  }

  // Next, we read the column names and display them.
  //
  // NOTICE: You must *always* read the column names even if
  //         you do not use them. This is so the connector can
  //         read the data out of the buffer. Row data follows the
  //         column data and thus must be read first.

  c = my_conn.get_columns();
  for (int i = 0; i < c->num_fields; i++) {
    Serial.print(c->fields[i]->name);
    if (i < c->num_fields - 1) {
      Serial.print(",");
    }
  }
  Serial.println();

  // Next, we use the get_next_row() iterator and read rows printing
  // the values returned until the get_next_row() returns NULL.

  int num_cols = c->num_fields;
  int rows = 0;
  do {
    r = my_conn.get_next_row();
    if (r) {
      rows++;
      for (int i = 0; i < num_cols; i++) {
        Serial.print(r->values[i]);
        if (i < num_cols - 1) {
          Serial.print(", ");
        }
      }
      Serial.println();

      // Note: we free the row read to free the memory allocated for it.
      // You should do this after you've processed the row.

      my_conn.free_row_buffer();
    }
  } while (r);
  Serial.print(rows);
  Serial.println(" rows in result.");

  // Finally, we are done so we free the column buffers

  my_conn.free_columns_buffer();
}

Arduino:1.6.5 (Windows 7), Scheda:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (1M SPIFFS)"

Forecast-Q.ino: In function 'void do_query(const char*)':
Forecast-Q:91: error: 'column_names' was not declared in this scope
Forecast-Q:91: error: 'c' was not declared in this scope
Forecast-Q:92: error: 'row_values' was not declared in this scope
Forecast-Q:92: error: 'r' was not declared in this scope
Forecast-Q:107: error: 'class Connector' has no member named 'get_columns'
Forecast-Q:122: error: 'class Connector' has no member named 'get_next_row'
Forecast-Q:136: error: 'class Connector' has no member named 'free_row_buffer'
Forecast-Q:144: error: 'class Connector' has no member named 'free_columns_buffer'
'column_names' was not declared in this scope

asetyde, 2015-10-22:
on example i think this is a key point #include <sha1.h> but i don't now how is ... sha crypto ?

asetyde, 2015-10-22:
https://github.com/Cathedrow/Cryptosuite for sha1.h but no resolve error

asetyde, 2015-10-22:
sha1 / sha256 library  http://forum.arduino.cc/index.php?action=dlattach;topic=119848.0;attach=95632

asetyde, 2015-10-22:
but also sha is not fully compatible ... some avr function etc ... "arduino portability bla bla bla"

asetyde, 2015-10-22:
at now this function  in mysql.cpp must cut or comment

boolean Connector::scramble_password(char *password, byte *pwd_hash) {..} 
sha1 issue

asetyde, 2015-10-22:
ok , creator Dot. Charles Bell write to me , many many thx , after some working now work and i ask @igrr  how add to our esp environment 

as i say this must be changed :   

[image]

you must take from this link manual and two libraries :  

put mysql library and sha1 library to arduino library folder

after this must be change in sha1 #include <avr/pgmspace.h> to #include <pgmspace.h>
also sha256 has some error but it's not used and i not touch it but need revision .

After this library worked :) now i test and i ll update post

asetyde, 2015-10-22:
( it can be uncommon also WITH SELECT to use example )

asetyde, 2015-10-23:
all seems be ok :) : :+1:

asetyde, 2015-12-22:
not working so good

asetyde, 2015-12-22:
https://github.com/esp8266/Arduino/issues/1233

darcicidade, 2016-06-20:
Hello, how are you?
I can get connection between Arduino and mysql with libraries below.
With these librarian it is possible to change and reading in the database.
No need to script in php.

#include <Ethernet.h>
#include <MySQL_Connection.h>
#include <MySQL_Cursor.h>
#include <avr/pgmspace.h>

Now ! I would like to do the same with Esp8266.

you get anything?

see you later.

f41ardu, 2016-07-22:
See my modifications. It run smooth with the latest lib whcih can be found on 
https://github.com/ChuckBell/MySQL_Connector_Arduino. 
All other examples have to be midified in the same manner. 
Don't forget to include the <> in the lib includes. They disapear while quoting code. 

f41ardu

`/*
  MySQL Connector/Arduino Example : basic insert

  This example demonstrates how to issue an INSERT query to store data in a
  table. For this, we will create a special database and table for testing.
  The following are the SQL commands you will need to run in order to setup
  your database for running this sketch.

  CREATE DATABASE test_arduino;
  CREATE TABLE test_arduino.hello_arduino (
    num integer primary key auto_increment,
    message char(40),
    recorded timestamp
  );

  Here we see one database and a table with three fields; a primary key that
  is an auto_increment, a string, and a timestamp. This will demonstrate how
  to save a date and time of when the row was inserted, which can help you
  determine when data was recorded or updated.

  INSTRUCTIONS FOR USE

  1) Create the database and table as shown above.
  2) Change the address of the server to the IP address of the MySQL server
  3) Change the user and password to a valid MySQL user and password
  4) Connect a USB cable to your Arduino
  5) Select the correct board and port
  6) Compile and upload the sketch to your Arduino
  7) Once uploaded, open Serial Monitor (use 115200 speed) and observe
  8) After the sketch has run for some time, open a mysql client and issue
     the command: "SELECT * FROM test_arduino.hello_arduino" to see the data
     recorded. Note the field values and how the database handles both the
     auto_increment and timestamp fields for us. You can clear the data with
     "DELETE FROM test_arduino.hello_arduino".

  Note: The MAC address can be anything so long as it is unique on your network.

  Created by: Dr. Charles A. Bell 
  Modified by: f41ardu for use with nodemcu
 
*/
#include  ESP8266WiFi.h 
#include  WiFiClient.h     
#include  MySQL_Connection.h 
#include  MySQL_Cursor.h 

IPAddress server_addr(192,168,1,34);  // IP of the MySQL *server* here
char user[] = "user";              // MySQL user login username
char password[] = "password";        // MySQL user login password

// WiFi card example
char ssid[] = "MYSSID";    // your SSID
char pass[] = "SSIDPASSWORD";       // your SSID Password

// Sample query
char INSERT_SQL[] = "INSERT INTO test_arduino.hello_arduino (message) VALUES ('Hello, Arduino!')";

WiFiClient client;       
MySQL_Connection conn((Client *)&client);

void setup() {
  Serial.begin(115200);
  // Begin WiFi section
  WiFi.begin(ssid, pass);
    // Wait for connection
  while ( WiFi.status() != WL_CONNECTED ) {
    delay ( 500 );
    Serial.print ( "." );
  }
  Serial.println ( "" );
  Serial.print ( "Connected to " );
  Serial.println ( ssid );
  Serial.print ( "IP address: " );
  Serial.println ( WiFi.localIP() );
  // End WiFi section
  Serial.println("DB  - Connecting...");
  while (conn.connect(server_addr, 3306, user, password) != true) {
    delay(500);
    Serial.print ( "." );
  }
}

void loop() {
  delay(2000);

  Serial.println("Recording data.");

  // Initiate the query class instance
  MySQL_Cursor *cur_mem = new MySQL_Cursor(&conn);
  // Execute the query
  cur_mem->execute(INSERT_SQL);
  // Note: since there are no results, we do not need to read any data
  // Deleting the cursor also frees up memory used
  delete cur_mem;
}
`

chadouming, 2016-07-22:
You can try this version too : https://github.com/chadouming/mysql-connector
. It has all the modification required to be used on esp8266.

darcicidade, 2016-07-23:
thank you so much.
Now I would like to know about SELECT.
I need to get values in different rows of the database.
I'm using (COLUMN_NAMES * my_conn.get_columns columns = ();)
But the command (delete my_conn;) not working.
Thank you

asetyde, 2016-07-25:
mysql connector is now included in arduino library browser , you can install it without problem and get periodically update. 
Use other version can not work so good .
Doc. Bell does extreme good work and it' oracle made , therefore i say all. 
Good bye this thread is close.

asetyde, 2016-07-25:
you can use this link , with all guidelines 
http://drcharlesbell.blogspot.it/2016/01/new-release-mysql-connectorarduino-110a.html

Also doctor Bell usually reply and give support for bus and other.

closed 2016-07-25 by asetyde

sachlj, 2017-02-18:
I use code with esp8266. 
`esp8266_mysql.ino:42: error: #include expects "FILENAME" or <FILENAME>
 #include ESP8266WiFi.h
esp8266_mysql.ino:43: error: #include expects "FILENAME" or <FILENAME>
 #include WiFiClient.h
esp8266_mysql.ino:44: error: #include expects "FILENAME" or <FILENAME>
 #include MySQL_Connection.h
esp8266_mysql.ino:45: error: #include expects "FILENAME" or <FILENAME>
 #include MySQL_Cursor.h
exit status 1
#include expects "FILENAME" or <FILENAME>
`
What file is missing? Sorry for my english.